RT-Thread/rt-thread #9143: [Bug] menuconfig is not working

nicholas-zww, 2024-07-06:
### RT-Thread Version

  master brach

### Hardware Type/Architectures

ART-Pi

### Develop Toolchain

Other

### Describe the bug

when using the latest env tool (https://github.com/RT-Thread/env) together with latest rt-thread master branch, menuconfig show "kconfig-mconf: not found"

nik@nik:~/Workbench/rt-thread/bsp/stm32/stm32h750-artpi$ menuconfig 
scons: Entering directory `/home/nik/Workbench/rt-thread/tools/kconfig-frontends'
Could not change directory to /home/nik/Workbench/rt-thread/tools/kconfig-frontends
scons: Reading SConscript files ...
Newlib version: 3.3.0
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
scons: `.' is up to date.
scons: done building targets.
sh: 1: /home/nik/Workbench/rt-thread/tools/kconfig-frontends/kconfig-mconf: not found
nik@nik:~/Workbench/rt-thread/bsp/stm32/stm32h750-artpi$ git log -1
commit d266ed6c432682de266ad27ffc057c2125c70fc7 (HEAD -> master, origin/master, origin/HEAD)
Author: Kai <60053077+kaidegit@users.noreply.github.com>
Date:   Sat Jul 6 16:05:35 2024 +0800

    [tool][cmake] Let cmake generator get more param from `rtconfig.py` (#9139)
    
    let cmake generator get more param from `rtconfig.py`
    
    * support getting c/cpp standard from c/cpp compilation flags
    * support getting post action and conv it to cmake variable
    * add a custom cmake file for user custom settings
    * support setting project name from `--project-name=` option

BernardXiong, 2024-07-07:
What's your host environment? You can try it with `scons --menuconfig` in Linux.

nicholas-zww, 2024-07-07:
host is ubuntu 24.04LTS.
Yes. I tied that, but still fail. refer to below 

nik@nik:~/Workbench/rt-thread/bsp/stm32/stm32h750-artpi$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 24.04 LTS
Release:    24.04
Codename:   noble
nik@nik:~/Workbench/rt-thread/bsp/stm32/stm32h750-artpi$ scons --menuconfig
scons: Reading SConscript files ...
ModuleNotFoundError: No module named 'menuconfig':
  File "/home/nik/Workbench/rt-thread/bsp/stm32/stm32h750-artpi/SConstruct", line 48:
    objs = PrepareBuilding(env, RTT_ROOT, has_libcpu=False)
  File "/home/nik/Workbench/rt-thread/tools/building.py", line 345:
    menuconfig(Rtt_Root)
  File "/home/nik/Workbench/rt-thread/tools/menukconfig.py", line 260:
    import menuconfig
nik@nik:~/Workbench/rt-thread/bsp/stm32/stm32h750-artpi$ echo $RTT_ROOT
/home/nik/Workbench/rt-thread

Later-Comer, 2024-07-07:
please try run `pip install kconfiglib`，then run `scons --menuconfig`

nicholas-zww, 2024-07-08:
Yes, it works now. Thanks

closed 2024-07-08 by nicholas-zww