RT-Thread/rt-thread #9161: [Bug] scons doesn 't work

unicornx, 2024-07-09:
### RT-Thread Version

master

### Hardware Type/Architectures

bsp/cvitek

### Develop Toolchain

Other

### Describe the bug

- 我原先工作在 v5.1.0 上，一切正常
- 后 checkout 到 master 并 pull 最新的版本，此时执行 scons --menuconfig，报错
- 执行 pip3 install -i https://pypi.tuna.tsinghua.edu.cn/simple kconfiglib 后可以工作。
- 切换回 v5.1.0, scons 又不工作。
- 执行 pip3 uninstall kconfiglib 卸载 kconfiglib 后 scons 还是不工作，并报如下错误：

```
$ scons 

scons: *** No SConstruct file found.
File "/usr/lib/python3/dist-packages/SCons/Script/Main.py", line 947, in _main
```

Later-Comer, 2024-07-09:
目前env版本针对不同的rtt版本支持情况有些残次不齐。

master版本只支持最新版本的[env master](https://github.com/RT-Thread/env/tree/master)，且建议使用[install_ubuntu.sh](https://github.com/RT-Thread/env/blob/master/install_ubuntu.sh)进行安装，否则需要手动安装pip install kconfiglib。其他版本建议使用老版本[env v1.5.x](https://github.com/RT-Thread/env/tree/v1.5.x)

因为master版本将原来老的kconfig-backend升级为python版本的kconfiglib，也就是说如果您使用老的env开发master必须手动安装kconfiglib。如果您使用老版本的env开发老版本的rtt，那么必须uninstall kconfiglib，否则会冲突

Later-Comer, 2024-07-09:
我这几天尽量把rtt仓库现有的分支都修改成支持kconfiglib

Later-Comer, 2024-07-09:
可以参考下：https://github.com/RT-Thread/rt-thread/issues/9141

unicornx, 2024-07-10:
> 很抱歉，kconfiglib的升级过程中，没有较全面的考虑对历史版本的兼容。
> 
> 我这几天尽量把rtt仓库现有的分支都修改成支持kconfiglib，但是tag估计需要重新发布

tag 的定义就是打上就不要动了，如果要改动，建议是增加小版本号，譬如 5.1.1，然后重新打tag，否则 tag 就失去意义了。

I have to say: 随意改动 tag 不是一个好的 release 习惯。

unicornx, 2024-07-10:
> 目前env版本针对不同的rtt版本支持情况有些残次不齐。
>
> master版本只支持最新版本的[env [...]

我主要在 linux/ubuntu 环境下工作（指开发 RTT），好像从来没用过 env 哎，我一直以为 env 是针对 windows 环境下的用户的呢。

Later-Comer, 2024-07-10:
> 我主要在 linux/ubuntu 环境下工作（指开发 RTT），好像从来没用过 env 哎，我一直以为 env 是针对 windows 环境下的用户的呢。

Ubuntu下工作，scons --menuconfig这个命令在rtt之前的老版本上是调用了kconfig-backend，但是在rtt master更换为python kconfiglib，但是由于老版本的rtt在tools目录下存在一些跟kconfiglib重名的python文件，因此如果开发老版本rtt那么必须uninstall kconfiglib，如果开发master版本必须安装kconfiglib。

如果你安装了最新版本的env master，那么使用menuconfig命令(env定制修改的一个工具)是可以兼容老旧版本的rtt的，不要使用scons --menuconfig（调用了部分python脚本和kconfiglib重名了）

windows下，由于有env-win这个存在，用户比较少使用scons --menuconfig，而是直接使用menuconfig.exe，这是一个定制的工具，因此可以兼容老旧版本

polarvid, 2024-07-10:
可以用 docker 建两个沙箱，一个用专门开发旧版本，一个专门开发 master 😂

polarvid, 2024-07-10:
> 如果你安装了最新版本的env master，那么使用menuconfig命令(env定制修改的一个工具)是可以兼容老旧版本的rtt的，不要使用scons --menuconfig（调用了部分python脚本和kconfiglib重名了）

@Later-Comer 既然如此，是否可以把 scons --menuconfig 重定向到 env？或者给一些提示信息。否则使用者很容易困惑。

Later-Comer, 2024-07-10:
是这样的情况，在rtt v5.1.0和之前的版本

- `scons --menuconfig`只能linux下使用，windows下是不能使用的，linux下运行`scons --menuconfig`先自动编译`tools\kconfig-backend`，然后os.system调用执行
- windows是通过env-windows定制的一个menuconfig.exe（本质是调用kconfig-backend的windows版本），因此windows用户基本上只是知道`menuconfig.exe`不知道`scons --menuconfig`
- 由于rt-thread/tools目录下有部分文件名和python kconfiglib重名，因此rt-thread v5.1.0之前版本跟kconfiglib冲突，想要执行scons --menuconfig必须卸载`pip uninstall kconfiglib`
- 由于历史遗留原因`scons --menuconfig`在linux环境下的确会下载env但是并不会安装依赖项（比如kconfiglib），且下载的是最新版本的env master，这个是有问题，rtt master会尽快修复

在的rt-thread master版本上，我们采用python kconfiglib替换了kconfig-backend：

- 统一了windows和linux的习惯，也就是都可以使用`scons --menuconfig`和`menuconfig`
- env master也同步修改，并推荐install_ubuntu.sh方式安装，这种方式安装会自动安装所有的pip依赖项（比如scons、kconfiglib等），启动env的时候会自动创建venv虚拟环境，并pip install env，此时windows会下自动生成menuconfig.exe，pkgs.exe，rt-env.exe等入口程序（**linux这部分还有修改到位**）。
- 由于windows用户多一些，因此我们优先升级了了env-windows到v2.0，这样windows用户使用v2.0能够编译rtt master，并能够大部分能够兼容rtt v5.1.0之前menuconfig，scons --pyconfig，但是无法兼容scons --menuconfig，因为tools下文件和kconfiglib库文件冲突
- linux下我们主要是在跑ci，目前master版本ci跑起来基本没什么问题，但是没有兼顾到部分linux用户在master和v5.1.0之前版本的切换，导致工具链失败

总体来说:

- 想要env 2.0全面兼容rtt v5.1.0及之前的版本，基本无法实现，因为env 2.0升级了python3，之前是python2，存在部分语法不兼容，这需要修改修改rtt/tools才能实现
- 在windows下env-windows v2.0能够部分兼容rtt v5.1.0及之前的版本，也能够无缝衔接之前的操作习惯
- 在linux下由于之前只是考虑ci，没有考虑用户操作，因此env还有修改到位，在rtt版本切换的时候会报kconfiglib库冲突或缺失，我尽快修改下
- 因此建议rtt v5.1.0及之前版本使用env 1.5.x，而rtt master版本必须使用env 2.0，如果linux环境下，切换到env 1.5.x必须pip uninstall kconfiglib（否则会和tools下py文件冲突），切换到env 2.0必须pip install kconfiglib。更好的解决方案是采用venv虚拟环境

Later-Comer, 2024-07-10:
> @Later-Comer 既然如此，是否可以把 scons --menuconfig 重定向到 env 呢？或者执行 scons --menuconfig 出错时丢一些有效的提示信息。否则使用者很容易困惑。

现在比较奇葩的是，scons --menuconfig是直接调用kconfiglib库，而env提供了一个定制的menuconfig（遗传自env-windows），他本质是将scons --menuconfig的功能复制了一次

其实我更倾向rtt的所有操作全面使用 scons --xxx的方式

Later-Comer, 2024-07-10:
> 看上去 env 是另一个配置的入口。那原先的 scons 的那套入口 （--menuconfig ）怎么和新的 env 入口融合起来感觉现在还做的有问题。
> 
> 特别是我原先只知道 scons 那套命令，现在又要学习 env 的命令，这个过度有什么好办法吗？
> 
> 其实我觉得最好只保留 scons 的入口，保持原有习惯不好吗？为啥有了 scons --menuconfig 还要有一个 Env 的 menuconfig。

我完全同意所有rtt才做都采用scons --xx这样的入口，由于env-win遗留menuconfig入口，同时也支持scons --menuconfig，不过现在在切换rtt版本的时候，linux环境下没有衔接好。另外env提供不仅仅是menuconfig，还有pkgs等多个操作，其实pkgs我是希望采用scons --pkgs方式。

想打是希望env独立成一个完整的工具，这个工具就可以集成到不同的ide里面，然后兼容各种ide对rtt进行操作。

mysterywolf, 2024-07-10:
scons --menuconfig 之前是Linux下的命令，menuconfig是windows下的命令，两年前在Linux下也可以直接使用menuconifg了，换句话说，scons --menuconfig 和 menuconfig是等同的。

BernardXiong, 2024-07-11:
如果是`scons --xx`的方式，也可以是完全集成到`rt-thread/tools`下了。但又希望后续env可以集成到不同的IDE里，所以两难了

Later-Comer, 2024-07-11:
> 如果是`scons --xx`的方式，也可以是完全集成到`rt-thread/tools`下了。但又希望后续env可以集成到不同的IDE里，所以两难了

其实scons --menuconfig可以完全调用env下的python，其中env是通用的，但是scons --menuconfig是从tools/building.py启动，这样可以兼容env的通用，同时还是在building.py里面定制rtt版本特殊的功能。