RT-Thread/rt-thread #9169: [Feature] rtt进入临界区直接粗暴的关掉全局中断，为何不像freRtos那样，只关闭受操作系统管理的中，而不关闭不受操作系统管理的中断

yu1999hub, 2024-07-10:
### Describe problem solved by the proposed feature

.global rt_hw_interrupt_disable
.type rt_hw_interrupt_disable, %function
rt_hw_interrupt_disable:
    MRS     r0, PRIMASK
    CPSID   I
    BX      LR

/*

yu1999hub, 2024-07-10:
[https://github.com/RT-Thread/rt-thread/blob/master/libcpu/arm/cortex-m4/context_gcc.S](url)

BernardXiong, 2024-07-12:
这个就是要做区分的了，一些可能也会和芯片相关（是否支持中断硬件抢占，以及位数是多少等）

yu1999hub, 2024-07-12:
rtthead有计划要做相关的支持吗

yu1999hub, 2024-07-12:
https://github.com/RT-Thread/rt-thread/issues/9169#issuecomment-2224258480
rtos

yu1999hub, 2024-07-12:
> 这个就是要做区分的了，一些可能也会和芯片相关（是否支持中断硬件抢占，以及位数是多少等）

rtthead 有计划支持可选择性的关闭中断吗

logeexpluoqi, 2024-07-15:
在串口数据高速收发中，921600波特率，1000Hz发送频率，serial_v2框架，非阻塞接收，一个包16-24个字节之间，经常遇到包接收不完整导致的丢包，这个会不会也是由于全局中断关闭造成的？

BernardXiong, 2024-07-21:
> > 这个就是要做区分的了，一些可能也会和芯片相关（是否支持中断硬件抢占，以及位数是多少等）
> 
> rtthead 有计划支持可选择性的关闭中断吗

可以给出方案出来，大家一起来review，讨论

QIANWC, 2025-02-20:
basepri issue提出已经几年了
https://github.com/RT-Thread/rt-thread/issues/5808#issuecomment-1146984762
https://github.com/RT-Thread/rt-thread/issues/7110

yu1999hub, 2025-06-13:
> 在串口数据高速收发中，921600波特率，1000Hz发送频率，serial_v2框架，非阻塞接收，一个包16-24个字节之间，经常遇到包接收不完整导致的丢包，这个会不会也是由于全局中断关闭造成的？

实测会出现丢中断的现象，在实时运动控制领域这是不可接受的

a1012112796, 2025-06-13:
题外话，我这最近调试 gd32h7 似乎遇到了相反的情况， 系统运行一段时间后，assert 出错了。coremark 打出的调用信息显示，似乎在定时器中断中调用了关终端保护的动作，但是依然触发了串口中断。就挺奇怪的...

[image]