RT-Thread/rt-thread #9209: [Bug] bsp/cvitek  aarch64 failed to build

unicornx, 2024-07-18:
### RT-Thread Version

master

### Hardware Type/Architectures

bsp/cvitek

### Develop Toolchain

GCC

### Describe the bug

尝试 build 主线上的 bsp/cvitek/cv18xx_aarch64/， 无法通过，遇到以下问题：

- 问题 1： bsp/cvitek/drivers/drv_gpio.c 编译失败：
```shell
/home/u/ws/duo/rt-thread/bsp/cvitek/drivers/drv_gpio.c: In function 'rt_hw_gpio_isr':
/home/u/ws/duo/rt-thread/bsp/cvitek/drivers/drv_gpio.c:274:22: error: 'SYS_GPIO_IRQ_BASE' undeclared (first use in this function)
  274 |     port = (irqno == SYS_GPIO_IRQ_BASE ? 4 : (irqno - GPIO_IRQ_BASE));
      |                      ^~~~~~~~~~~~~~~~~
/home/u/ws/duo/rt-thread/bsp/cvitek/drivers/drv_gpio.c:274:22: note: each undeclared identifier is reported only once for each function it appears in
/home/u/ws/duo/rt-thread/bsp/cvitek/drivers/drv_gpio.c:274:55: error: 'GPIO_IRQ_BASE' undeclared (first use in this function); did you mean 'UART_IRQ_BASE'?
  274 |     port = (irqno == SYS_GPIO_IRQ_BASE ? 4 : (irqno - GPIO_IRQ_BASE));
      |                                                       ^~~~~~~~~~~~~
      |                                                       UART_IRQ_BASE
/home/u/ws/duo/rt-thread/bsp/cvitek/drivers/drv_gpio.c: In function 'rt_hw_gpio_init':
/home/u/ws/duo/rt-thread/bsp/cvitek/drivers/drv_gpio.c:309:29: error: 'GPIO_IRQ_BASE' undeclared (first use in this function); did you mean 'UART_IRQ_BASE'?
  309 |     rt_hw_interrupt_install(GPIO_IRQ_BASE + (no), rt_hw_gpio_isr, RT_NULL, "gpio");    \
      |                             ^~~~~~~~~~~~~
/home/u/ws/duo/rt-thread/bsp/cvitek/drivers/drv_gpio.c:312:5: note: in expansion of macro 'INT_INSTALL_GPIO_DEVICE'
  312 |     INT_INSTALL_GPIO_DEVICE(0);
      |     ^~~~~~~~~~~~~~~~~~~~~~~
/home/u/ws/duo/rt-thread/bsp/cvitek/drivers/drv_gpio.c:317:29: error: 'SYS_GPIO_IRQ_BASE' undeclared (first use in this function)
  317 |     rt_hw_interrupt_install(SYS_GPIO_IRQ_BASE, rt_hw_gpio_isr, RT_NULL, "gpio");
      |                             ^~~~~~~~~~~~~~~~~
scons: *** [build/drivers/drv_gpio.o] Error 1
scons: building terminated because of errors.
```

- 问题 2：bsp/cvitek/drivers/drv_pinmux.c 编译失败：
```shell
CC build/drivers/drv_pinmux.o
/home/u/ws/duo/rt-thread/bsp/cvitek/drivers/drv_pinmux.c:477:2: error: #error "Unsupported SOC type!"
  477 | #error "Unsupported SOC type!"
      |  ^~~~~
/home/u/ws/duo/rt-thread/bsp/cvitek/drivers/drv_pinmux.c: In function 'pinmux_get_index':
/home/u/ws/duo/rt-thread/bsp/cvitek/drivers/drv_pinmux.c:485:15: error: 'pin_selects_array' undeclared (first use in this function)
  485 |         p = &(pin_selects_array[pin_index][i]);
      |               ^~~~~~~~~~~~~~~~~
/home/u/ws/duo/rt-thread/bsp/cvitek/drivers/drv_pinmux.c:485:15: note: each undeclared identifier is reported only once for each function it appears in
/home/u/ws/duo/rt-thread/bsp/cvitek/drivers/drv_pinmux.c: In function 'pinmux_config':
/home/u/ws/duo/rt-thread/bsp/cvitek/drivers/drv_pinmux.c:517:40: error: 'pinmux_array' undeclared (first use in this function)
  517 |     for (index = 0; index < ARRAY_SIZE(pinmux_array); index++) {
      |                                        ^~~~~~~~~~~~
/home/u/ws/duo/rt-thread/bsp/cvitek/drivers/drv_pinmux.c:23:36: note: in definition of macro 'ARRAY_SIZE'
   23 | #define ARRAY_SIZE(ar)     (sizeof(ar)/sizeof(ar[0]))
      |                                    ^~
scons: *** [build/drivers/drv_pinmux.o] Error 1
scons: building terminated because of errors.
```

问题 3：链接时提示 rt_fdt_commit_memregion_early 找不到：
```shell
LINK rtthread.elf
/home/u/ws/bin/rtt-aarch64-gcc/arm-gnu-toolchain-13.3.rel1-x86_64-aarch64-none-elf/bin/../lib/gcc/aarch64-none-elf/13.3.1/../../../../aarch64-none-elf/bin/ld: warning: rtthread.elf has a LOAD segment with RWX permissions
/home/u/ws/bin/rtt-aarch64-gcc/arm-gnu-toolchain-13.3.rel1-x86_64-aarch64-none-elf/bin/../lib/gcc/aarch64-none-elf/13.3.1/../../../../aarch64-none-elf/bin/ld: build/board/board.o: in function `rt_hw_board_init':
/home/u/ws/duo/rt-thread/bsp/cvitek/cv18xx_aarch64/board/board.c:82:(.text+0x38): undefined reference to `rt_fdt_commit_memregion_early'
/home/u/ws/duo/rt-thread/bsp/cvitek/cv18xx_aarch64/board/board.c:82:(.text+0x38): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `rt_fdt_commit_memregion_early'
```

问题 4：bsp/cvitek/drivers/drv_uart.c 编译告警
```shell
CC build/drivers/drv_uart.o
/home/u/ws/duo/rt-thread/bsp/cvitek/drivers/drv_uart.c: In function 'rt_hw_uart_init':
/home/u/ws/duo/rt-thread/bsp/cvitek/drivers/drv_uart.c:367:29: warning: variable 'config' set but not used [-Wunused-but-set-variable]
  367 |     struct serial_configure config = RT_SERIAL_CONFIG_DEFAULT;
      |                             ^~~~~~
/home/u/ws/duo/rt-thread/bsp/cvitek/drivers/drv_uart.c:366:28: warning: unused variable 'uart' [-Wunused-variable]
  366 |     struct hw_uart_device* uart;
      |                            ^~~~
/home/u/ws/duo/rt-thread/bsp/cvitek/drivers/drv_uart.c: At top level:
/home/u/ws/duo/rt-thread/bsp/cvitek/drivers/drv_uart.c:205:13: warning: 'rt_hw_uart_isr' defined but not used [-Wunused-function]
  205 | static void rt_hw_uart_isr(int irqno, void *param)
      |             ^~~~~~~~~~~~~~
/home/u/ws/duo/rt-thread/bsp/cvitek/drivers/drv_uart.c:197:33: warning: '_uart_ops' defined but not used [-Wunused-const-variable=]
  197 | static const struct rt_uart_ops _uart_ops =
      |                                 ^~~~~~~~~
```

unicornx, 2024-07-18:
简单的分析如下：

问题 1
----

09a0e4c5f8 引入了 SYS_GPIO_IRQ_BASE 和 GPIO_IRQ_BASE， 但只对 riscv 的 bsp 做了配置，遗漏了 arm64 的

问题 2
----

a63c07fd75 引入了 drv_pinmux，目前其实并没有支持 arm64，所以原来的设计上存在缺陷，需要考虑一下如何支持 arm64。

问题 3：

master 上貌似已经不支持 rt_fdt_commit_memregion_early' 了，需要看一下应该怎么处理

问题 4：

未分析。TBD

Z8MAN8, 2024-07-20:
## 问题3
对比了 9c359fd4c82db644894d7a6e51810e570d5bd6ca 和现在最新 libcpu/aarch64/common/setup.c 中 'rt_hw_common_setup' 实现，bsp/cvitek/cv18xx_aarch64/board/board.c 中的`rt_fdt_commit_memregion_early' 部分是否可以直接删除

liYony, 2024-07-21:
> ## 问题3
> 对比了 [9c359fd](https://github.com/RT-Thread/rt-thread/commit/9c359fd4c82db644894d7a6e51810e570d5bd6ca) 和现在最新 libcpu/aarch64/common/setup.c 中 'rt_hw_common_setup' 实现，bsp/cvitek/cv18xx_aarch64/board/board.c 中的`rt_fdt_commit_memregion_early' 部分是否可以直接删除

直接删除即可，参考qemu-aarch64

liYony, 2024-07-21:
> 简单的分析如下：
> 
> ## 问题 1
> [09a0e4c](https://github.com/RT-Thread/rt-thread/commit/09a0e4c5f83fab24348e95d52fba7180a82abe12) 引入了 SYS_GPIO_IRQ_BASE 和 GPIO_IRQ_BASE， 但只对 riscv 的 bsp 做了配置，遗漏了 arm64 的
> 
> 已经单独提单 #9217 跟踪。
> 
> ## 问题 2
> [a63c07f](https://github.com/RT-Thread/rt-thread/commit/a63c07fd75ec9632e063646f043db3bfdad2aa49) 引入了 drv_pinmux，目前其实并没有支持 arm64，所以原来的设计上存在缺陷，需要考虑一下如何支持 arm64。
> 
> 已经单独提单 #9211 跟踪。
> 
> ## 问题 3：
> master 上貌似已经不支持 `rt_fdt_commit_memregion_early' 了，需要看一下应该怎么处理
> 
> ## 问题 4：
> 已经单独提单 #9220 跟踪。

针对问题1和问题2，我看是宏的问题，因为rv和arm这两个bsp都是支持设备树，要不要考虑直接从设备树去解析这些信息呢？？我觉得没必要用宏

unicornx, 2024-07-22:
> 针对问题1和问题2，我看是宏的问题，因为rv和arm这两个bsp都是支持设备树，要不要考虑直接从设备树去解析这些信息呢？？我觉得没必要用宏

[image]
[image]
[image]
[image]

unicornx, 2024-07-29:
所有拆分的问题已经解决，关闭此 issue。

unicornx, 2024-07-29:
@Z8MAN8 FYI.

closed 2024-07-29 by unicornx

Z8MAN8, 2024-07-29:
> @Z8MAN8 FYI.

收到，感谢！