RT-Thread/rt-thread #9214: [Bug] USB UAC1 is not working with smartphone.

barfiii, 2024-07-18:
### RT-Thread Version

3.1.3

### Hardware Type/Architectures

Rockchip

### Develop Toolchain

GCC

### Describe the bug

The USB UAC1 device is not functioning properly with Smartphones, while it is working as expected with a PC. The USB UAC1 device is not being detected as earphones after connecting it to a mobile device. However, if you manually restart the device after connecting it to the smartphone, it starts working. This behavior suggests a potential compatibility issue or a software-related problem. What could be the problem?

BernardXiong, 2024-07-21:
There is no RK bsp on RTT 3.1.x series.

SKrishna004, 2024-07-21:
> There is no RK bsp on RTT 3.1.x series.
> 

I'm talking about this **bsp/rockchip/rk2108**

barfiii, 2024-07-21:
I'm talking about **bsp/rockchip/rk2108**

BernardXiong, 2024-07-24:
Hi Barfiii,
There is no usb driver under rockchip bsp. Therefore, this issue is closed.

closed 2024-07-24 by BernardXiong