RT-Thread/rt-thread #9215: [Bug] bsp/cvitek documentation needs cleanup

unicornx, 2024-07-18:
### RT-Thread Version

master

### Hardware Type/Architectures

bsp/cvitek

### Develop Toolchain

GCC

### Describe the bug

目前 bsp/cvitek 下文档感觉有点乱，目前有四份 readme：
- `bsp/cvitek/README.md` （中文）
- `bsp/cvitek/c906_little/README.md`（中文/英文）
- `bsp/cvitek/cv18xx_aarch64/README.md`（中文）
- `bsp/cvitek/cv18xx_risc-v/README.md`（中文/英文）

我建议清理一下。清理的方案我想了一下，可以有以下两种方式：

- 方案 A:  每个 bsp 一份，即去掉 `bsp/cvitek/README.md`，或者将其内容和 `bsp/cvitek/cv18xx_risc-v/README.md` 合并

- 方案 B : 去掉 `bsp/cvitek/c906_little/README.md` 和 `bsp/cvitek/cv18xx_risc-v/README.md`，全部合并入 `bsp/cvitek/README.md`，同时在 `bsp/cvitek/README.md` 中加一个链接指向 `bsp/cvitek/cv18xx_aarch64/README.md`

我比较倾向于方案 B，因为对于 C906B + C906L 的方式，单独描述小核，或者单独大核都是没有意义的，现在的文档中也可以看到这一点，在小核的 readme 中也会介绍大核的烧写，反之亦然。而 `bsp/cvitek/README.md` 综合介绍了大核和小核，读起来更通顺。对于 ARM + C906B 的方式，描述相对独立，可以保留 `bsp/cvitek/cv18xx_aarch64/README.md`，然后在 `bsp/cvitek/README.md` 中加一个链接指向 `bsp/cvitek/cv18xx_aarch64/README.md` 即可。

unicornx, 2024-07-18:
@polarvid 请 review 我的修改方案，谢谢。

polarvid, 2024-07-19:
@unicornx 

个人比较赞同方案 B。道理正如 PR 描述信息里的一样。另外官方选择的也是 RISC-V 大小核作为默认启动方式，ARM64 大核相当于特殊情况，需要调整 boot 引脚电平。所以文档默认就是 riscv 大小核方案也是与板子一致的。

liYony, 2024-07-19:
> @unicornx
> 
> 个人比较赞同方案 B。道理正如 PR 描述信息里的一样。另外官方选择的也是 RISC-V 大小核作为默认启动方式。ARM64 大核相当于特殊情况，需要调整 boot 引脚电平。所以文档默认就是 riscv 大小核方案也是与板子一致的。

+1 赞同

flyingcys, 2024-07-21:
> @unicornx
> 
> 个人比较赞同方案 B。道理正如 PR 描述信息里的一样。另外官方选择的也是 RISC-V 大小核作为默认启动方式。ARM64 大核相当于特殊情况，需要调整 boot 引脚电平。所以文档默认就是 riscv 大小核方案也是与板子一致的。
+1 赞同

PR #9233 (2024-07-22): bsp: cvitek: improve documentation
Fixed #9215 

## 拉取/合并请求描述：(PR description)

[

#### 为什么提交这份PR (why to submit this PR)

详见 commit 描述

#### 你的解决方案是什么 (what is your solution)

详见 commit 描述

注意，这次修改没有提供英文版本，是否需要？

#### 请提供验证的bsp和config (provide the config and bsp) 

- BSP: bsp/cvitek

- .config: 文档改进，和代码无关

- action: 文档改进，和代码无关

]
closed 2024-07-22 by mysterywolf