RT-Thread/rt-thread #9223: [Bug] Could not compile the rt-thread\bsp\stm32\stm32f429-fire-challenger in

mz1874, 2024-07-20:
### RT-Thread Version

5.10.0

### Hardware Type/Architectures

stm32f429-fire-challenger

### Develop Toolchain

MDK Keil

### Describe the bug

The example of stm32f429-fire-challenger can not be compiled in 5.10.0, and , I already tried the solution of this link https://github.com/RT-Thread/rt-thread/issues/8948  and this blog https://blog.csdn.net/dujunqiu/article/details/139826672?spm=1001.2014.3001.5501    it still not woking . I hope the problem can be soloved officially. If can not i have to downgrade the version to 5.01

BernardXiong, 2024-07-25:
The officially way is to update project by `scons --target=mdk -s` or `scons --target=mkd5 -s`.

mz1874, 2024-07-30:
from 阿里邮箱 iPhone
thank u i solved it
 ------------------Original Mail ------------------
Sender:Bernard Xiong ***@***.***>
Send Date:Thu Jul 25 09:27:27 2024
Recipients:RT-Thread/rt-thread ***@***.***>
CC:WangChong ***@***.***>, Author ***@***.***>
Subject:Re: [RT-Thread/rt-thread] [Bug] Could not compile the rt-thread\bsp\stm32\stm32f429-fire-challenger in (Issue #9223)

The officially way is to update project by scons --target=mdk -s or scons --target=mkd5 -s.