RT-Thread/rt-thread #9232: [Bug] bsp/qemu-virt64-riscv/SConstruct 中的 stack_size 问题

unicornx, 2024-07-22:
### RT-Thread Version

master

### Hardware Type/Architectures

bsp/qemu-virt64-riscv

### Develop Toolchain

GCC

### Describe the bug

`bsp/qemu-virt64-riscv/SConstruct`

```python
stack_lds = open('link_stacksize.lds', 'w')
if GetDepend('__STACKSIZE__'): stack_size = GetDepend('__STACKSIZE__')
stack_lds.write('__STACKSIZE__ = %d;\n' % stack_size)
stack_lds.close()
```   

这里  `stack_size = GetDepend('__STACKSIZE__')` 是有何用意？GetDepend 返回值是 False

不过看上去 `GetDepend('__STACKSIZE__')` 一直返回的是 False，倒也没做什么。但是这个引入一个问题就是 `__STACKSIZE__` 的配置修改在 `bsp/qemu-virt64-riscv/link_stacksize.lds` 中不会起作用。

这个类似问题在很多 risv64 的项目中都存在，估计是抄过去的。

```shell
$ find -name "SConstruct" | xargs grep -wn "GetDepend('__STACKSIZE__')"
./allwinner/d1/SConstruct:58:if GetDepend('__STACKSIZE__'): stack_size = GetDepend('__STACKSIZE__')
./allwinner/d1s/SConstruct:96:if GetDepend('__STACKSIZE__'): stack_size = GetDepend('__STACKSIZE__')
./k210/SConstruct:31:if GetDepend('__STACKSIZE__'): stack_size = GetDepend('__STACKSIZE__')
./qemu-virt64-riscv/SConstruct:37:if GetDepend('__STACKSIZE__'): stack_size = GetDepend('__STACKSIZE__')
./bouffalo_lab/bl808/d0/SConstruct:47:if GetDepend('__STACKSIZE__'): stack_size = GetDepend('__STACKSIZE__')
./cvitek/cv18xx_risc-v/SConstruct:30:if GetDepend('__STACKSIZE__'): stack_size = GetDepend('__STACKSIZE__')
./juicevm/SConstruct:31:if GetDepend('__STACKSIZE__'): stack_size = GetDepend('__STACKSIZE__')
```

polarvid, 2024-07-22:
`__STACKSIZE__` 是 bsp 下面的 Kconfig 里面定义的配置项目。这里就是尝试读取这个配置，然后写入链接脚本里面。

[image]

unicornx, 2024-07-22:
> `__STACKSIZE__` 是 bsp 下面的 Kconfig 里面定义的配置项目。这里就是尝试读取这个配置，然后写入链接脚本里面。 [...]

代码能看明白，但是逻辑我觉得是写错了。

polarvid, 2024-07-23:
> 代码能看明白，但是逻辑我觉得是写错了。

我本地上看是没什么问题的。验证如下。

1. 修改配置 stack size 为 32KB

[image]

2. 查看读取的配置 size

[image]

3. 验证链接脚本的 stack size

[image]

unicornx, 2024-07-23:
sorry，我发现描述错误，这个问题只存在于 bsp/cvitek/cv18xx_riscv 中。已经另外提了一个 issue #9237 来跟踪。

关闭此 issue。

closed 2024-07-24 by unicornx