RT-Thread/rt-thread #9243: [Bug] bsp/cvitek/cv18xx_aarch64 无法 boot 进入控制台界面

unicornx, 2024-07-24:
### RT-Thread Version

master

### Hardware Type/Architectures

bsp/cvitek/cv18xx_aarch64

### Develop Toolchain

GCC

### Describe the bug

目前 master 上的代码 duo 的 aarch64 大核无法启动进入控制台界面，只能打印出 RT flag 就 hold 住了。

bisect 看了一下，最近一个可以 work 的 commit 是 `ae6a3287e6 Add psoc62、61 config.`
加入 `754c59a411 [Feature] DFS mount auto by kernel parameters (#8989)` 后就会出现该现象。

继续跟踪了一下，发现卡住是因为 `components/drivers/core/mnt.c` 文件中的 `rootfs_mnt_init()` 进入了死循环。

```c
static int rootfs_mnt_init(void)
{
......
        for (; rootdelay || rootwait; --timeout)
        {
            if (!rootwait && timeout == 0)
            {
                LOG_E("Wait for /dev/%s init time out", dev);

                /*
                 * We don't return at once because the device driver may init OK
                 * when we break from this point, might as well give it another
                 * try.
                 */
                break;
            }

            if (rt_device_find(dev))
            {
                break;
            }

            rt_thread_mdelay(1);
        }
......
}
INIT_ENV_EXPORT(rootfs_mnt_init);
```

这个函数，包括整个文件都是  `754c59a411 [Feature] DFS mount auto by kernel parameters (#8989)` 新引入的。所以怀疑是引入整个 feature 后 bsp/cvitek/cv18xx_aarch 需要做一些对应的改动但没有做导致。

所以想问问具体缺失了什么？

GuEe-GUI, 2024-07-24:
设备树启动参数没有 rootwait，这里是不会死等的：

https://github.com/RT-Thread/rt-thread/blob/8dc166b16c8ee9c0d219ad1bd6d633a8f8f8c95d/components/drivers/core/mnt.c#L73-L86

unicornx, 2024-07-24:
@liYony FYI, 是不是设备树上有什么要改动？

unicornx, 2024-07-24:
> 设备树启动参数没有 rootwait，这里是不会死等的：
> 
> https://github.com/RT-Thread/rt-thread/blob/8dc166b16c8ee9c0d219ad1bd6d633a8f8f8c95d/components/drivers/core/mnt.c#L73-L86

@GuEe-GUI 请问您的意思是我们的 DTS 里少了 rootwait 这个参数吗，这些参数是必需的吗？

另外有关 `rootfs_mnt_init·()` 函数的代码我有个疑问。

我发现由于我们提供的 dtb 中没有给出 rootwait 和 rootdelay，

```c
rootwait = bootargs_select("rootwait", 0);
```
这个调用会返回一个空字符串(rootwait 的值也不是 NULL)。 这导致 for 循环中无法满足 break 的条件。

请问这是不是一个bug？

GuEe-GUI, 2024-07-24:
> @GuEe-GUI 请问您的意思是我们的 DTS 里少了 rootwait 这个参数吗，这些参数是必需的吗？
>
> 另外有关 `rootfs_mnt_init·()` 函数的代码我有个疑问。
>
> 我发现由于我们提供的 dtb 中没有给出 rootwait 和 rootdelay，
>
> ```c
> rootwait = bootargs_select("rootwait", 0);
> ``` [...]

如果返回了非 NULL 的空字符串，那就是有 BUG。

GuEe-GUI, 2024-07-24:
> @GuEe-GUI 请问您的意思是我们的 DTS 里少了 rootwait 这个参数吗，这些参数是必需的吗？
>
> 另外有关 `rootfs_mnt_init·()` 函数的代码我有个疑问。
>
> 我发现由于我们提供的 dtb 中没有给出 rootwait 和 rootdelay，
>
> ```c
> rootwait = bootargs_select("rootwait", 0);
> ``` [...]

如果 bootargs 没有这个参数，依然返回了了非 NULL 的空字符串，那就是有 BUG。

unicornx, 2024-07-24:
> > > 设备树启动参数没有 rootwait，这里是不会死等的：
> > > https://github.com/RT-Thread/rt-thread/blob/8dc166b16c8ee9c0d219ad1bd6d633a8f8f8c95d/components/drivers/core/mnt.c#L73-L86
> > 
> > 
> > @GuEe-GUI 请问您的意思是我们的 DTS 里少了 rootwait 这个参数吗，这些参数是必需的吗？
> > 另外有关 `rootfs_mnt_init·()` 函数的代码我有个疑问。
> > 我发现由于我们提供的 dtb 中没有给出 rootwait 和 rootdelay，
> > ```c
> > rootwait = bootargs_select("rootwait", 0);
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 这个调用会返回一个空字符串(rootwait 的值也不是 NULL)。 这导致 for 循环中无法满足 break 的条件。
> > 请问这是不是一个bug？
> 
> 如果 bootargs 没有这个参数，依然返回了了非 NULL 的空字符串，那就是有 BUG，不妨把这个字符串打印出来看看。

打印出来是空的，rootwait 的值也打印了，和初始值（目前随机）不同，感觉是分配了一个内存了，但内容为空。

unicornx, 2024-07-24:
[sg2002_milkv_duo256m_arm_sd.txt](https://github.com/user-attachments/files/16359610/sg2002_milkv_duo256m_arm_sd.txt)

上传反汇编的 dts 文件，txt 后缀改成 dts 即可

GuEe-GUI, 2024-07-24:
> 打印出来是空的，rootwait 的值也打印了，和初始值（目前随机）不同，感觉是分配了一个内存了，但内容为空。

抱歉，没描述清楚，因为 rootwait 后面已经没有字符串了（如果 bootargs 有这个参数的话），那 *rootwait 就是一个开头为 '\0' 的字符串，也就是说，只要 select 的 key 存在，返回值就肯定不是空，而是这个 key 后面的字符串（空格之前的）。

GuEe-GUI, 2024-07-24:
> [sg2002_milkv_duo256m_arm_sd.txt](https://github.com/user-attachments/files/16359610/sg2002_milkv_duo256m_arm_sd.txt)
> 
> 上传反汇编的 dts 文件，txt 后缀改成 dts 即可
> 
> @GuEe-GUI 供参考

请问这个是 Bootloader 运行时修改的设备树还是在 Linux source 下构建的设备树？

unicornx, 2024-07-24:
> 抱歉，没描述清楚，因为 rootwait 后面已经没有字符串了（如果 bootargs 有这个参数的话），那 *rootwait 就是一个开头为 '\0' 的字符串，也就是说，只要 select 的 key 存在，返回值就肯定不是空，而是这个 key 后面的字符串（空格之前的）。

> > [sg2002_milkv_duo256m_arm_sd.txt](https://github.com/user-attachments/files/16359610/sg2002_milkv_duo256m_arm_sd.txt)
> > 上传反汇编的 dts 文件，txt 后缀改成 dts 即可
> > @GuEe-GUI 供参考
> 
> 请问这个是 Bootloader 运行时修改的设备树还是在 Linux source 下构建的设备树？

@liYony 请回复一下。谢谢

unicornx, 2024-07-24:
> 抱歉，没描述清楚，因为 rootwait 后面已经没有字符串了（如果 bootargs 有这个参数的话），那 *rootwait 就是一个开头为 '\0' 的字符串，也就是说，只要 select 的 key 存在，返回值就肯定不是空，而是这个 key 后面的字符串（空格之前的）。

没看懂您这里说的意思，我们的 dts 里没有 bootargs 呢。

GuEe-GUI, 2024-07-24:
> > > [sg2002_milkv_duo256m_arm_sd.txt](https://github.com/user-attachments/files/16359610/sg2002_milkv_duo256m_arm_sd.txt)
> > > 上传反汇编的 dts 文件，txt 后缀改成 dts 即可
> > > @GuEe-GUI 供参考
> > 
> > 
> > 请问这个是 Bootloader 运行时修改的设备树还是在 Linux source 下构建的设备树？
> 
> @liYony 请回复一下。谢谢

最直接的做法，启动后把 bootargs 都打印出来就可以，如果串口可以早期打印：

```c
    const char *bootargs;
    struct rt_ofw_node *ofw_chosen_node = rt_ofw_find_node_by_path("/chosen");

    if (!rt_ofw_prop_read_string(ofw_chosen_node, "bootargs", &bootargs))
    {
        rt_kputs("Command line: ");
        // 可能会很长被截断
        rt_kputs(bootargs);
        rt_kputs("\n");
    }
```

unicornx, 2024-07-24:
> > > > [sg2002_milkv_duo256m_arm_sd.txt](https://github.com/user-attachments/files/16359610/sg2002_milkv_duo256m_arm_sd.txt)
> > > > 上传反汇编的 dts 文件，txt 后缀改成 dts 即可
> > > > @GuEe-GUI 供参考
> > > 
> > > 
> > > 请问这个是 Bootloader 运行时修改的设备树还是在 Linux source 下构建的设备树？
> > 
> > 
> > @liYony 请回复一下。谢谢
> 
> 最直接的做法，启动后把 bootargs 都打印出来就可以，如果串口可以早期打印：
> 
> ```c
>     const char *bootargs;
>     struct rt_ofw_node *ofw_chosen_node = rt_ofw_find_node_by_path("/chosen");
> 
>     if (!rt_ofw_prop_read_string(ofw_chosen_node, "bootargs", &bootargs))
>     {
>         rt_kputs("Command line: ");
>         // 可能会很长被截断，就不使用 printf 了
>         rt_kputs(bootargs);
>         rt_kputs("\n");
>     }
> ```

@GuEe-GUI 我把这段代码就加在 rootfs_mnt_init 的入口处得到如下打印：

```shell
 \ | /
- RT -     Thread Operating System
 / | \     5.2.0 build Jul 24 2024 17:04:11
 2006 - 2024 Copyright by RT-Thread team
Command line: root=/dev/mmcblk0p2 rootwait rw console=ttyS0,115200 earlycon=sbi riscv.fwsz=0x80000 loglevel=9
```

GuEe-GUI, 2024-07-24:
> @GuEe-GUI 看上去是 bootloader 加了这个 rootwait？
> 
> 所以我们这里会一直等待直到 `/dev/mmcblk0p2` 创建成功才会退出？
> 
> ```c
> for（）
> {
> ......
>   if (rt_device_find(dev))
>    {
>        break;
>    }
> ```

是的，如果你们暂时没完成这个驱动，也不方便修改 Bootloader 对设备树的修改配置的话（原本的设计是 bootargs 说了算），可以暂时不使用这段自动挂载的代码，注释就好了。

unicornx, 2024-07-24:
@liYony 请看一下这个问题该怎么解决最简单呢

我想过是否可以关掉 RT_USING_DFS（因为 mnt.c 是否参与编译由这个控制） ，但似乎这个选项被很多依赖，至少我在 menuconfig 里无法关掉它。

Z8MAN8, 2024-07-24:
我可以负责继续推进

unicornx, 2024-07-25:
目前打算采用暂时的 workaround 解法，因为duo的 arm 大核其实目前用户很少，功能实现也比较简单，我们这只是在整理 riscv 的代码时涉及一些公共部分，先是发现 arm 的 bsp 编译不过，然后就是编译过了起不来，现在发现问题是在这里。但我理解如果后面 riscv 如果也要上设备树，则这个问题是不是还会冒出来。但目前来说 duo 只有 arm 的 bsp 用了设备树，所以遇到了这个问题

临时的解法可以伪造一个名字叫 `/dev/mmcblk0p2` 的设备, 循环就会退出了, 结束之后反注册这个设备.

临时解法局限在arn的bsp里就好。

参考实现代码如下：
```c
#include <rtthread.h>
#include <rtdevice.h>

static struct rt_device *pseudo_mmcblk;

static int pseudo_mmcblk_setup(void)
{
    pseudo_mmcblk = rt_calloc(1, sizeof(*pseudo_mmcblk));

    RT_ASSERT(pseudo_mmcblk != RT_NULL);

    pseudo_mmcblk->type = RT_Device_Class_Graphic;

    return (int)rt_device_register(pseudo_mmcblk, "/dev/mmcblk0p2", RT_DEVICE_FLAG_DEACTIVATE);
}
INIT_BOARD_EXPORT(pseudo_mmcblk_setup);

static int pseudo_mmcblk_remove(void)
{
    if (pseudo_mmcblk)
    {
        return (int)rt_device_unregister(pseudo_mmcblk);
    }

    return 0;
}
INIT_FS_EXPORT(pseudo_mmcblk_remove);
```

closed 2024-07-25 by mysterywolf