RT-Thread/rt-thread #9279: [Bug] FAL头文件可能会污染ulog的LOG_TAG

kaidegit, 2024-08-07:
### RT-Thread Version

master

### Hardware Type/Architectures

h750

### Develop Toolchain

GCC

### Describe the bug

引入或间接引入会导致log_d等api的标签受到污染

在程序中定义LOG_TAG后引入ulog后再引入fal，或是先引入fal再定义LOG_TAG引入ulog，均会导致log_d等小写api的TAG变成FAL

[image]
[image]
[image]

wdfk-prog, 2024-08-07:
调换一下顺序,这一块放到最后
```c
#define LOG_TAG              "example"
#define LOG_LVL              LOG_LVL_DBG
#include <ulog.h>
```

kaidegit, 2024-08-07:
唔似乎要求在包含fal之前的头文件都没有包含过ulog，否则最下面的包含ulog直接就被忽略了，也不是很友好

BernardXiong, 2024-08-08:
或者这类是否有好的解决办法，大家可以多想想

mysterywolf, 2024-11-24:
https://github.com/RT-Thread/rt-thread/pull/9690

closed 2024-11-24 by mysterywolf