RT-Thread/rt-thread #9281: [Bug] bsp/cvitek cv18xx_aarch64 build error

Z8MAN8, 2024-08-07:
### RT-Thread Version

master

### Hardware Type/Architectures

bsp/cvitek

### Develop Toolchain

GCC

### Describe the bug

尝试 build 主线上的 bsp/cvitek/cv18xx_aarch64， 无法通过，遇到以下问题：
components/libc/compilers/common/ctime.c 编译失败：
```shell
/home/csc/Project/rt-thread/components/libc/compilers/common/ctime.c:607:20: warning: implicit declaration of function '_control_rtc' [-Wimplicit-function-declaration]
  607 |             return _control_rtc(RT_DEVICE_CTRL_RTC_GET_TIMERES, res);
      |                    ^~~~~~~~~~~~
/home/csc/Project/rt-thread/components/libc/compilers/common/ctime.c:607:33: error: 'RT_DEVICE_CTRL_RTC_GET_TIMERES' undeclared (first use in this function); did you mean 'RT_DEVICE_CTRL_GET_INT'?
  607 |             return _control_rtc(RT_DEVICE_CTRL_RTC_GET_TIMERES, res);
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                 RT_DEVICE_CTRL_GET_INT
/home/csc/Project/rt-thread/components/libc/compilers/common/ctime.c:607:33: note: each undeclared identifier is reported only once for each function it appears in
/home/csc/Project/rt-thread/components/libc/compilers/common/ctime.c: In function 'clock_gettime':
/home/csc/Project/rt-thread/components/libc/compilers/common/ctime.c:638:33: error: 'RT_DEVICE_CTRL_RTC_GET_TIMESPEC' undeclared (first use in this function); did you mean 'RT_DEVICE_CTRL_GET_INT'?
  638 |             return _control_rtc(RT_DEVICE_CTRL_RTC_GET_TIMESPEC, tp);
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                 RT_DEVICE_CTRL_GET_INT
/home/csc/Project/rt-thread/components/libc/compilers/common/ctime.c: In function 'clock_nanosleep':
/home/csc/Project/rt-thread/components/libc/compilers/common/ctime.c:680:36: error: 'RT_DEVICE_CTRL_RTC_GET_TIMESPEC' undeclared (first use in this function); did you mean 'RT_DEVICE_CTRL_GET_INT'?
  680 |                 err = _control_rtc(RT_DEVICE_CTRL_RTC_GET_TIMESPEC, &ts);
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                    RT_DEVICE_CTRL_GET_INT
/home/csc/Project/rt-thread/components/libc/compilers/common/ctime.c: In function 'clock_settime':
/home/csc/Project/rt-thread/components/libc/compilers/common/ctime.c:731:33: error: 'RT_DEVICE_CTRL_RTC_SET_TIMESPEC' undeclared (first use in this function); did you mean 'RT_DEVICE_CTRL_SET_INT'?
  731 |             return _control_rtc(RT_DEVICE_CTRL_RTC_SET_TIMESPEC, (void *)tp);
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                 RT_DEVICE_CTRL_SET_INT
/home/csc/Project/rt-thread/components/libc/compilers/common/ctime.c: In function 'timer_settime':
/home/csc/Project/rt-thread/components/libc/compilers/common/ctime.c:1197:36: error: 'RT_DEVICE_CTRL_RTC_GET_TIMESPEC' undeclared (first use in this function); did you mean 'RT_DEVICE_CTRL_GET_INT'?
 1197 |                 err = _control_rtc(RT_DEVICE_CTRL_RTC_GET_TIMESPEC, &ts);
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                    RT_DEVICE_CTRL_GET_INT
scons: *** [build/kernel/components/libc/compilers/common/ctime.o] Error 1
scons: building terminated because of errors.
``` 
通过版本回退，定位到 47cd52d162248eed0b8a5d6a1d04c870874ddae8 引入该问题。
简单分析如下：该改动中，通过 RT_USING_RTC 宏判断控制了 rtdevice.h 头文件的引用，但在 components/libc/compilers/common/ctime.c 中没有处理干净依赖于该头文件的相关代码，导致编译错误。这部分代码也应该需要添加对应的宏判断控制编译。

Z8MAN8, 2024-08-07:
<details><summary>Details</summary>
<p>

</p>
</details>

Z8MAN8, 2024-08-07:
@unicornx

unicornx, 2024-08-07:
我怀疑是 bsp/cvitek/cv18xx_aarch 的 配置中没有使能 RT_USING_RTC 导致

riscv 的大小核都使能了 RT_USING_RTC，所以 ctime.c 都是可以编译的

还要再看看。不确定为啥以前是好的

Z8MAN8, 2024-08-07:
> 我怀疑是 bsp/cvitek/cv18xx_aarch 的 配置中没有使能 RT_USING_RTC 导致
> 
> riscv 的大小核都使能了 RT_USING_RTC，所以 ctime.c 都是可以编译的
> 
> 还要再看看。不确定为啥以前是好的，以及是不是 ctime.c 本身的问题。

我仔细看了发现，小核并没有使能 RT_USING_RTC 也能正常构建，大核riscv目前是使能了 RT_USING_RTC 且能正常编译，但若大核riscv不使能 RT_USING_RTC 会出现同样的编译错误。小核不使能 RT_USING_RTC 也能正常编译的原因是 #if defined(RT_USING_POSIX_TIMER) && defined(RT_USING_KTIME) 不满足该条件，存在问题的目标代码并不会参与编译，但大核满足。并且在 ctime.c 查看 _control_rtc 的定义和其他调用都使用 #ifdef RT_USING_RTC 进行了判断，但 #if defined(RT_USING_POSIX_TIMER) && defined(RT_USING_KTIME) 作用域中的 _control_rtc 调用没有做相关宏判断，导致可能出现依赖缺失的问题，这部分可能是 47cd52d 的改动者遗漏了，因为改动之间是没有对RTC相关头文件的引用进行宏判断，因此之前无论是否定义 RT_USING_RTC 都不会出现依赖缺失的问题。

Z8MAN8, 2024-08-07:
在 https://github.com/RT-Thread/rt-thread/commit/47cd52d162248eed0b8a5d6a1d04c870874ddae8 之前，ctimer.c 是默认引用了 rtdevice.h 但如果没有使能 RT_USING_DEVICE 就会缺失这个头文件，rtc 相关的依赖也会缺失，所以作者通过预编译开关处理了一下，但出问题的这个地方应该是遗漏了

closed 2024-08-11 by Rbb666