RT-Thread/rt-thread #9315: [Bug] config options are undefined

unicornx, 2024-08-17:
### RT-Thread Version

master

### Hardware Type/Architectures

bsp/cvitek

### Develop Toolchain

Other

### Describe the bug

`bsp/cvitek/cv18xx_risc-v/Kconfig`

```
choice BSP_ROOTFS_TYPE
    ......
    config BSP_ROOTFS_TYPE_CROMFS
       ......
        select PKG_USING_ZLIB
        select PKG_USING_ZLIB_LATEST_VERSION
endchoice
```
其中 `PKG_USING_ZLIB` 和 `PKG_USING_ZLIB_LATEST_VERSION` 在 RTT 中没有找到定义，感觉应该去掉

unicornx, 2024-08-17:
这个选项是 cfa3ecfa9eee0b580d59e5a24cb374a72a5fc34e 这个 commit 引入的，对应 PR：#9229

unicornx, 2024-08-17:
@polarvid 请看一下这是不是一个 issue？

polarvid, 2024-08-20:
@unicornx 这个是在在外部软件包中，需要通过 env 工具获取。配置位于 https://github.com/RT-Thread/packages/blob/9e06f23adcb2167656d4af1e3d1cb46331b2d2b0/misc/zlib/Kconfig#L7 。

BernardXiong, 2024-08-25:
这个就是RTT的软件包机制的，然后用menuconfig后，应该用其中的搜索功能就能够找到

unicornx, 2024-08-25:
ok, 那不是问题，我先关闭这个 issue

closed 2024-08-25 by unicornx