RT-Thread/rt-thread #9316: [Feature] 统一 bsp/cvitek 的 riscv 的大小核的项目构建

unicornx, 2024-08-17:
### Describe problem solved by the proposed feature

目前 bsp/cvitek 的 riscv 部分的大小核的构建完全遵循了两套机制。小核 c906_little 用的是 `CPU ='rv64'`, 大核 cv18xx_riscv 用的是 `CPU ='c906'`。可是据我了解 cv18xx 芯片上的 riscv 大小核本质上没有什么大的区别，除了中断号配置有些不同外。

所以我建议是否可以考虑优化一下代码，将两个 bsp 的项目构建统一起来，同时考虑到目前大核的开发工作更成熟，所以都统一为大核的处理方式。

请大家看一下是否可行以及是否有更好的建议？

unicornx, 2024-08-17:
@BernardXiong @flyingcys  @polarvid @Z8MAN8 各位在 duo 上参与贡献的比较多, 请从历史和经验上给出你们的建议。谢谢

flyingcys, 2024-08-19:
确认有mmu的话，可以合并

BernardXiong, 2024-08-25:
大小核这边，是否都可以用[mlibc](https://github.com/plctlab/mlibc)？如果可以的话，是完全可以的。

BernardXiong, 2024-08-25:
另外建议是尽快启用设备树，然后相同rv64核的芯片（应该指的都是大核），只需要提供不同的设备树，用同一个内核也ok的。

如果小核也带mmu，则大小核用一个内核（带不同的设备树）也是可以的

unicornx, 2024-08-25:
> 大小核这边，是否都可以用[mlibc](https://github.com/plctlab/mlibc)？如果可以的话，是完全可以的。

mlibc 在 RTT 上是什么情况，还不太了解，谁能解释一下？

unicornx, 2024-08-26:
> 另外建议是尽快启用设备树，然后相同rv64核的芯片（应该指的都是大核），只需要提供不同的设备树，用同一个内核也ok的。
> 
> 如果小核也带mmu，则大小核用一个内核（带不同的设备树）也是可以的

设备树应该是另一个话题了，我的想法是先基于目前状态先把大小核统一了再考虑引入新的特性

unicornx, 2024-09-07:
BSP_UART_IRQ_BASE		30	60	44
BSP_I2C_IRQ_BASE		32	N/A	49
BSP_TIMER_IRQ_BASE		55	N/A	79

IRQ_MAX_NR				61	N/A	101

BSP_GPIO_IRQ_BASE		41	76	60
BSP_SYS_GPIO_IRQ_BASE	47	86	70