RT-Thread/rt-thread #9321: [Feature] 希望能在tools目录添加一个autoenvsave.py方便命令行用户

heyuanjie87, 2024-08-20:
### Describe problem solved by the proposed feature

命令行用户往往需要设置RTT_EXEC_PATH这个环境变量，对于多人协作项目来说不能在代码里把他改成固定值，调用脚本设置需要每次进终端输入一串字符这太麻烦了，给短短人生增加了上亿次额外输入不太划算，所以我想把需要配置的环境变量保存起来，下次只需要执行scons就可以了。

### Describe your preferred solution

下面是我写好的代码autoenvsave.py

```
import os
import configparser

def getenv(name, defualt=''):
    value = ''

    conf = configparser.ConfigParser()
    conf.read('env.ini')

    osenv = os.getenv(name)
    if osenv:
        value = osenv
        if conf.get('env', name, fallback='') != osenv:
            if not conf.has_section('env'):
                conf.add_section('env')
            conf.set('env', name, value)
            with open('env.ini', 'w') as configfile:
                conf.write(configfile)
                configfile.close()
    else:
        value = conf.get('env', name, fallback='')

    if not value:
        value = defualt

    return value

```

rtconfig.py
```
RTT_ROOT = autoenvsave.getenv('RTT_ROOT', r'../../..')

if autoenvsave.getenv('RTT_EXEC_PATH'):
    EXEC_PATH = autoenvsave.getenv('RTT_EXEC_PATH')
```

BernardXiong, 2024-08-21:
用新版本env，安装了sdk后会自动查找工具链

ComerLater, 2024-08-21:
目前的版本是根据rtt_exec_path环境变量来确定工具链的路径，需要使用不同工具链的时候，的确需要要么修改环境变量，要么scons --exec-path指定路径，或者修改rtconfig.py，但是cross-tool也是通过环境变量或者scons --cross-tool方式来确定的。

一种比较好的思路，就是scons --useconfig=xxx.config的方式来指定编译特定的配置，xxx.config可以通过Kconfig配置保存下来，在xxx.config可以有cross-tool，甚至rtt-exec-path等，如果没有指定则使用环境变量的

heyuanjie87, 2024-08-21:
我这个方法就是把它存进文件里而且还不用保存到kconfig(多人协同时各自配置不同)

ComerLater, 2024-08-26:
> env 2.0比较少人使用吗？可以通过sdk来安装工具链

windows大部分兄弟都是使用env-win，这个开箱即用的工具，基本很少关注到sdk的用法

BernardXiong, 2024-09-09:
Due to the env script, this issue has been closed.

closed 2024-09-09 by BernardXiong