RT-Thread/rt-thread #9366: [Feature] 64 位架构适配 32 位指针 ABI | Using ILP32D ABI on 64-bit architecture

polarvid, 2024-08-29:
### Describe problem solved by the proposed feature

## 描述

#9194 在 QEMU 平台引入了 ILP32D 支持，同时也引入了内核对该 ABI 的支持需求。但内核、组件众多代码仍然隐式依赖、混用 intptr_t 与 RT-Thread 的架构关联类型 rt_base_t。

https://github.com/RT-Thread/rt-thread/blob/8249e71bcccffc8d4e96b32178f1e40863656ccf/include/rttypes.h#L36

如注释中清晰表达 —— `Nbit CPU related data type`，rt_base_t 直接与架构相关。这个类型的实际意义是需要一个专用类型来存储来自系统寄存器等对象的值。

而 `intptr_t` 来自标准 C，意义为 `integer type capable of holding a pointer`。为了充分发挥 ILP32D ABI 的优势，该类型自然是 32-bit 位长。

因此两种类型在这种场景下完全不能混淆。这导致了使用该 ABI 的工具链后，产生如下问题

1. 内核编译期产生大量警告（指针与不可比整型的隐式类型转换）
2. 内核中常使用 void * 存储物理地址。但是在 ILP32D 下，指针类型并不一定能表示所有的可用物理地址范围

BernardXiong, 2024-08-30:
ilp32d具体是怎么一回事？指针大小是？long的大小是？或者也包括size_t的大小是？

polarvid, 2024-08-30:
> ilp32d具体是怎么一回事？指针大小是？long的大小是？或者也包括size_t的大小是？

PR https://github.com/RT-Thread/rt-thread/pull/9361 中做了部分修复。此 ABI 下，`int`, `long`, `void *` 位长都是 32-bit 。

额外，size_t 应该是考虑到虚拟地址 32bit 空间，也调整为 32 bits 了。因为 size_t 设计意图是 sizeof 的返回类型。而既然地址空间（指针长度）都不能超过 32 bits，它定义为 32 bits 就是最合理的。