RT-Thread/rt-thread #9375: [Feature] 多人开发时编译器安装路径不一样导致rtconfig.h不方便提交的解决方案

sulfurandcu, 2024-09-02:
### Describe problem solved by the proposed feature

目前编译器路径配置是在rtconfig.h文件中修改的，多人开发时，每个人的编译器安装路径都不一样，修改rtconfig.h这个文件时就必须避免提交EXEC_PATH=""这行代码，很不方便。

### Describe your preferred solution

可以在rtconfig.h中添加如下代码，在首次编译时自动生成rtconfig.ini文件，并提示开发者填入自己电脑上的编译器路径。
```
import configparser

# toolchains path
cfg_path = r'rtconfig.ini'
if not os.path.exists(cfg_path):
    file = open(cfg_path, "w")
    file.write("# do not commit this file to the repository.\n")
    file.write("[COMPILER] # gcc, iar, mdk\n")
    file.write("COMPILER = gcc\n")
    file.write("GCC_PATH = replace this line with your arm-none-eabi-gcc path. (i.e. d:\\arm-none-eabi-10.3-2021.10\\bin) (do not commit this file to repository)\n")
    file.write("IAR_PATH =\n")
    file.write("MDK_PATH =\n")
    file.close()
    print('error: rtconfig.ini does not exist. but it should have been generated automatically, you need to fill in some informations.')
    exit()
cfg = configparser.ConfigParser()
cfg.read(cfg_path)
CROSS_TOOL = cfg["COMPILER"]["COMPILER"]
gcc_path = cfg["COMPILER"]["GCC_PATH"]
iar_path = cfg["COMPILER"]["IAR_PATH"]
mdk_path = cfg["COMPILER"]["MDK_PATH"]
```

然后将EXEC_PATH的配置改为：
```
if  CROSS_TOOL == 'gcc':
    PLATFORM    = 'gcc'
    EXEC_PATH   = gcc_path
elif CROSS_TOOL == 'iar':
    PLATFORM    = 'iccarm'
    EXEC_PATH   = iar_path
elif CROSS_TOOL == 'mdk':
    CROSS_TOOL  = 'keil'
    PLATFORM    = 'armcc'
    EXEC_PATH   = mdk_path
```

rtconfig.ini example:
```
# do not commit this file to repository.
[COMPILER] # gcc, iar, mdk
COMPILER = gcc
GCC_PATH = d:\arm-none-eabi-10.3-2021.10\bin
IAR_PATH =
MDK_PATH =
```

sakumisu, 2024-09-02:
错了，EXEC_PATH 是使用 env传入的，比如export EXEC_PATH = RTT_EXEC_PATH

supperthomas, 2024-09-03:
我觉得不用路径，直接用path路径，默认命令行中加这个命令比较方便一些。

sulfurandcu, 2024-09-03:
嗯，确实配置PATH全局生效更方便些，不过如果有个别工程有特殊需要的话倒是可以用这个来单独配置。

BernardXiong, 2024-09-04:
用env脚本方式，然后激活其中的venv环境，用env通过sdk方式安装工具链，然后就可以不用设置环境变量，不用改rtconfig.py文件了

BernardXiong, 2024-09-04:
可以看这篇文章中提及的

https://club.rt-thread.org/ask/article/eb77a696b86e6104.html

BernardXiong, 2024-09-09:
Due to the env script, this issue has been closed. And no setting in rtconfig.h but rtconfig.py.