RT-Thread/rt-thread #9397: [Bug] double free issue

SoftSec-Tech, 2024-09-05:
### RT-Thread Version

1cc0fb87dfc7bf348a7634690e6c550c75cd46f4

### Hardware Type/Architectures

stm32

### Develop Toolchain

GCC

### Describe the bug

1. There is double free issue in line 770 of https://github.com/RT-Thread/rt-thread/blob/master/components/finsh/msh.c.
2. This issue was detected by Static Analysis tool - StatiCode,
[image]
As in the screenshot, in step 38, call closedir(dir) and free the memory;  And then in step 39, call closeddir(dir) and free memory again., which caused double free issue.

ChinoUkaegbu, 2024-09-07:
Hi I'd like to work on this but I don't think there's a double free issue. It seems like the `closedir(dir)` on [line 770](https://github.com/RT-Thread/rt-thread/blob/66dd03f6c26caa15adb174e36f11544f60d3dadf/components/finsh/msh.c#L770) is freeing the variable defined in [line 631](https://github.com/RT-Thread/rt-thread/blob/66dd03f6c26caa15adb174e36f11544f60d3dadf/components/finsh/msh.c#L631) while the `closedir(dir)` on [line 761](https://github.com/RT-Thread/rt-thread/blob/66dd03f6c26caa15adb174e36f11544f60d3dadf/components/finsh/msh.c#L761) is freeing the variable defined in [line 759](https://github.com/RT-Thread/rt-thread/blob/66dd03f6c26caa15adb174e36f11544f60d3dadf/components/finsh/msh.c#L759). If that's the case then I believe one of the variables may need to be renamed.

BernardXiong, 2024-09-07:
> Hi I'd like to work on this but I don't think there's a double free issue. It seems like the `closedir(dir)` on [line [...]

Yes, you are right.

mysterywolf, 2024-09-25:
fixed in https://github.com/RT-Thread/rt-thread/pull/9464

closed 2024-09-25 by mysterywolf