RT-Thread/rt-thread #9407: [Feature] 瑞芯微RK3588J

frankzflyward, 2024-09-09:
### Describe problem solved by the proposed feature

瑞芯微RK3588J/RK3588的BSP支持

BernardXiong, 2024-09-09:
欢迎PR，提交及完善rockchip的BSP

GuEe-GUI, 2024-09-10:
rockchip 是目前主流的国产芯片，大概率会支持的，可以把你们对 3588 的应用需求描述出来方便社区开发人员把握重心

frankzflyward, 2024-09-11:
基于瑞芯微RK3588的主板运行RT-Thread上的需求

1. 项目背景
我们正在开发一款基于瑞芯微RK3588主板的实时控制系统，计划在该平台上运行RT-Thread操作系统。该系统将用于高性能工业自动化/无人机控制/机器人应用，涉及多种传感器和通信接口的集成。为了确保系统的高实时性和可靠性，我们希望RT-Thread社区能够支持并优化以下硬件接口和软件功能。

2. 硬件接口需求

2.1 RS-422接口

- 接口类型: RS-422
- 通信速率: 最高支持10 Mbps
- 设备数量: 预计连接10-20个RS-422设备
- 使用场景: RS-422接口将用于连接工业传感器，进行高速数据采集和实时控制。系统要求能够处理高频率数据传输，且需要可靠的通信驱动支持。

2.2 CAN接口

- 接口类型: CAN 2.0B / CAN FD
- 通信速率: 支持最高1 Mbps（CAN 2.0B）或8 Mbps（CAN FD）
- 节点数量: 系统最多连接8个CAN节点
- 使用场景: CAN总线用于多个节点之间的实时通信，如电机控制器和传感器模块的协调工作。希望RT-Thread能够提供稳定的CAN驱动支持，并兼容常用的CAN协议栈（如CANopen或J1939）。

2.3 相机通讯接口

- 接口类型: CameraLink / SDI / MIPI CSI-2 / USB 3.0
- 相机类型: 高分辨率工业相机，支持实时图像采集
- 图像要求: 1080p@60fps或更高，低延迟数据传输
- 使用场景: 相机接口将用于实时图像采集和处理，应用于机器视觉或导航系统中。需要确保图像数据的低延迟传输，并能与其他传感器数据同步处理。

3. 软件功能需求

3.1 实时性要求
实时性: 系统对实时性有严格要求，尤其是在传感器数据处理和控制信号发送方面。希望RT-Thread能够支持硬实时调度，并在多任务并发时保持稳定的响应时间。

3.2 协议栈支持

- RS-422协议: 系统将使用Modbus RTU协议在RS-422上进行通信。希望RT-Thread提供完整的Modbus协议栈支持，并能与RS-422接口无缝集成。
- CAN协议: 系统将基于CANopen（或其他指定协议）进行设备间通信。希望RT-Thread社区能提供或优化现有的CANopen协议栈，以确保与硬件接口的兼容性和可靠性。

3.3 设备驱动需求

- GPIO: 用于通用输入输出控制，希望能提供灵活的配置选项和中断处理功能。
- UART: 除RS-422外，还需支持标准UART通信，适用于低速调试和其他串行设备。

4. 应用场景

本系统将应用于以下场景：

- 工业自动化控制: 系统将用于控制多个工业传感器和执行器，需实现实时数据采集、处理和反馈。
- 无人机控制: 系统将应用于无人机的飞行控制，涉及多传感器数据融合和实时导航控制。
- 机器人导航: 系统将用于机器人导航，需实时处理视觉和惯性数据，提供精准的位姿估计和路径规划。

5. 性能和稳定性要求

- 系统性能: 预期系统处理器负载不超过80%，内存使用不超过70%，确保系统在高负载下依然能够稳定运行。
- 系统稳定性: 系统需支持长期连续运行，要求在极端环境下依然保持高稳定性。希望RT-Thread社区能够提供内存管理和错误处理方面的优化，确保系统的可靠性。

6. 社区协作

我们希望与RT-Thread社区紧密合作，共同开发和优化上述需求中的各项功能。我们愿意提供详细的硬件文档和现有代码，并参与驱动调试和系统优化工作。期待与社区开发者们的积极合作，共同推进RT-Thread在瑞芯微RK3588平台上的应用发展。

GuEe-GUI, 2025-04-10:
看上去社区版本只能提供上述的基本驱动支持，剩下的相机、RS模式、CAN协议栈这些高级货就不好说了呢

FixJA, 2025-04-22:
瑞芯微的RT-Thread支持是他们自己原厂维护的版本驱动支持已经很全面了，你说的这些驱动需要联系原厂拿SDK才有