RT-Thread/rt-thread #9454: [Bug] 瑞芯微rk3568 bsp failed to build 编译报错

Inglam, 2024-09-16:
### RT-Thread Version

master

### Hardware Type/Architectures

bsp/rk3568

### Develop Toolchain

GCC

### Describe the bug

参照瑞芯微RK3568的说明（https://github.com/RT-Thread/rt-thread/blob/master/bsp/rockchip/rk3568/README_zh.md）进行编译，运行`scons`命令后有如下报错：
`scons: Reading SConscript files ...
Newlib version: 4.1.0
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
LINK rtthread.elf
f:/瑞芯微 资料/gcc-arm-10.3-2021.07-mingw-w64-i686-aarch64-none-elf/bin/../lib/gcc/aarch64-none-elf/10.3.1/../../../../aarch64-none-elf/bin/ld.exe: build\kernel\libcpu\aarch64\common\setup.o: in function `rt_hw_common_setup':
F:\瑞芯微 资料\rt-thread-master\libcpu\aarch64\common/setup.c:233: undefined reference to `rt_memblock_reserve_memory'
F:\瑞芯微 资料\rt-thread-master\libcpu\aarch64\common/setup.c:233:(.text+0x3bc): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `rt_memblock_reserve_memory'
f:/瑞芯微 资料/gcc-arm-10.3-2021.07-mingw-w64-i686-aarch64-none-elf/bin/../lib/gcc/aarch64-none-elf/10.3.1/../../../../aarch64-none-elf/bin/ld.exe: F:\瑞芯微 资料\rt-thread-master\libcpu\aarch64\common/setup.c:234: undefined reference to `rt_memblock_reserve_memory'
F:\瑞芯微 资料\rt-thread-master\libcpu\aarch64\common/setup.c:234:(.text+0x3d4): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `rt_memblock_reserve_memory'
f:/瑞芯微 资料/gcc-arm-10.3-2021.07-mingw-w64-i686-aarch64-none-elf/bin/../lib/gcc/aarch64-none-elf/10.3.1/../../../../aarch64-none-elf/bin/ld.exe: F:\瑞芯微 资料\rt-thread-master\libcpu\aarch64\common/setup.c:235: undefined reference to `rt_memblock_reserve_memory'
F:\瑞芯微 资料\rt-thread-master\libcpu\aarch64\common/setup.c:235:(.text+0x3ec): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `rt_memblock_reserve_memory'
f:/瑞芯微 资料/gcc-arm-10.3-2021.07-mingw-w64-i686-aarch64-none-elf/bin/../lib/gcc/aarch64-none-elf/10.3.1/../../../../aarch64-none-elf/bin/ld.exe: F:\瑞芯微 资料\rt-thread-master\libcpu\aarch64\common/setup.c:236: undefined reference to `rt_memblock_reserve_memory'
F:\瑞芯微 资料\rt-thread-master\libcpu\aarch64\common/setup.c:236:(.text+0x404): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `rt_memblock_reserve_memory'
f:/瑞芯微 资料/gcc-arm-10.3-2021.07-mingw-w64-i686-aarch64-none-elf/bin/../lib/gcc/aarch64-none-elf/10.3.1/../../../../aarch64-none-elf/bin/ld.exe: F:\瑞芯微 资料\rt-thread-master\libcpu\aarch64\common/setup.c:270: undefined reference to `rt_memblock_setup_memory_environment'
F:\瑞芯微 资料\rt-thread-master\libcpu\aarch64\common/setup.c:270:(.text+0x574): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `rt_memblock_setup_memory_environment'
f:/瑞芯微 资料/gcc-arm-10.3-2021.07-mingw-w64-i686-aarch64-none-elf/bin/../lib/gcc/aarch64-none-elf/10.3.1/../../../../aarch64-none-elf/bin/ld.exe: build\kernel\components\drivers\ofw\fdt.o: in function `fdt_reserved_memory_reg':
F:\瑞芯微 资料\rt-thread-master\components\drivers\ofw/fdt.c:250: undefined reference to `rt_memblock_reserve_memory'
F:\瑞芯微 资料\rt-thread-master\components\drivers\ofw/fdt.c:250:(.text+0x894): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `rt_memblock_reserve_memory'
f:/瑞芯微 资料/gcc-arm-10.3-2021.07-mingw-w64-i686-aarch64-none-elf/bin/../lib/gcc/aarch64-none-elf/10.3.1/../../../../aarch64-none-elf/bin/ld.exe: build\kernel\components\drivers\ofw\fdt.o: in function `fdt_scan_memory':
F:\瑞芯微 资料\rt-thread-master\components\drivers\ofw/fdt.c:316: undefined reference to `rt_memblock_reserve_memory'
F:\瑞芯微 资料\rt-thread-master\components\drivers\ofw/fdt.c:316:(.text+0xab4): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `rt_memblock_reserve_memory'
f:/瑞芯微 资料/gcc-arm-10.3-2021.07-mingw-w64-i686-aarch64-none-elf/bin/../lib/gcc/aarch64-none-elf/10.3.1/../../../../aarch64-none-elf/bin/ld.exe: F:\瑞芯微 资料\rt-thread-master\components\drivers\ofw/fdt.c:357: undefined reference to `rt_memblock_add_memory'
F:\瑞芯微 资料\rt-thread-master\components\drivers\ofw/fdt.c:357:(.text+0xc7c): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `rt_memblock_add_memory'
f:/瑞芯微 资料/gcc-arm-10.3-2021.07-mingw-w64-i686-aarch64-none-elf/bin/../lib/gcc/aarch64-none-elf/10.3.1/../../../../aarch64-none-elf/bin/ld.exe: build\kernel\components\drivers\ofw\fdt.o: in function `fdt_scan_initrd':
F:\瑞芯微 资料\rt-thread-master\components\drivers\ofw/fdt.c:474: undefined reference to `rt_memblock_reserve_memory'
F:\瑞芯微 资料\rt-thread-master\components\drivers\ofw/fdt.c:474:(.text+0x116c): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `rt_memblock_reserve_memory'
Memory region         Used Size  Region Size  %age Used
collect2.exe: error: ld returned 1 exit status
scons: *** [rtthread.elf] Error 1
scons: building terminated because of errors.`

BernardXiong, 2024-09-16:
可以运行下menuconfig，让它从新生成下配置头文件，然后再进行编译

supperthomas, 2024-09-16:
请参考： https://github.com/RT-Thread/rt-thread/actions/runs/10887672974/job/30210444570

frankzflyward, 2024-09-17:
> 可以运行下menuconfig，让它从新生成下配置头文件，然后再进行编译

感谢熊先生。其实最开始scons还会报找不到变量的一些错，我往rtconfig.h里面手动define了一些量，然后按你现在的提示使用memconfig打开了全部memory management设置后，编译成功了。

> 请参考： https://github.com/RT-Thread/rt-thread/actions/runs/10887672974/job/30210444570

感谢指路。参照log看出，原来使用的GNU toolchain是https://github.com/RT-Thread/toolchains-ci/releases/download/v1.6/gcc-arm-10.2-2020.11-x86_64-aarch64-none-elf.tar.xz，而我之前使用的是https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-a/downloads/链接里最新的2021版本。换为2020.11版本后不需要修改rtconfig.h和memconfig也可以编译通过。

closed 2024-09-17 by supperthomas