RT-Thread/rt-thread #9479: [Bug] [ymodem] last frame's data_sz issue. (soh, stx)

sulfurandcu, 2024-09-27:
### RT-Thread Version

trunk

### Hardware Type/Architectures

all

### Develop Toolchain

Microsoft VScode

### Describe the bug

when ymodem receiving last frame in function _rym_do_fin(), 
in my opinion:
- if code is SOH, it should receive another `_RYM_SOH_PKG_SZ-1` long data,
- if code is STX, it should receive another `_RYM_STX_PKG_SZ-1` long data,

but now it will receive `_RYM_SOH_PKG_SZ-1` no matter what code is.
is there a bug or is the design like this?

[image]

mysterywolf, 2024-10-03:
Could you submit a PR please?

sulfurandcu, 2024-10-08:
> Could you submit a PR please?

see #9510

closed 2024-10-08 by sulfurandcu