RT-Thread/rt-thread #9521: [Feature] riscv: common64: ASID 的初始化过程优化 | Merge asid init into setup of early page table

polarvid, 2024-10-11:
### Describe problem solved by the proposed feature

关联 https://github.com/RT-Thread/rt-thread/pull/9511#discussion_r1796327187

https://github.com/RT-Thread/rt-thread/blob/906d1caea6f82dcb45dc1925842e89b76c01bbfd/libcpu/risc-v/common64/asid.c#L30-L51

这里重置 satp 怪怪的。因为正常是两个启动阶段，做两次 satp 设置。第一次是初始映射，使用大粒度 pte，为了给初始化过程配置最基础的虚存属性（比如 cache 属性），并进入线性地址空间。第二次是按照虚存配置的映射（可能从静态/动态配置中读取运行时虚存布局），按照应用、驱动的需求配置内核地址空间。

这里在中间又重置了一次 satp，并不是特别必要。或许是可以把 ASID 初始化操作合并到 `rt_hw_mem_setup_early` （第一次的初始映射）中。

zhangjing0303, 2024-10-11:
目前`rt_hw_asid_init`中通过向satp寄存器的bit[59:44]写1以确定ASIDLEN的值，此函数最好在MMU页表未使能前调用，所以把 ASID 初始化操作合并到` rt_hw_mem_setup_early`的开始处比较合理些。

zhangjing0303, 2024-10-11:
再就是`commn64`中通过`ARCH_USING_ASID`来决定是否使用ASID,是否可以通过`rt_hw_asid_init`中的ASID_BITS值确定当前cpu是否支持ASID，如果支持，则默认在rt_hw_aspace_switch中使用asid功能。

closed 2026-10-17 by Rbb666