RT-Thread/rt-thread #9526: [Bug] packages\wiznet-latest\src\wiz_af_inet.c(35): error:  #393: pointer to incomplete class type is not allowed

zrz4066, 2024-10-11:
### RT-Thread Version

rt-thread 5.x

### Hardware Type/Architectures

stm32f103-hw100k-ibox

### Develop Toolchain

MDK Keil

### Describe the bug

keil 5.x Compiler error
packages\wiznet-latest\src\wiz_af_inet.c(29): warning:  #231-D: declaration is not visible outside of function
  static int wiz_poll(struct dfs_fd *file, struct rt_pollreq *req)
packages\wiznet-latest\src\wiz_af_inet.c(35): error:  #393: pointer to incomplete class type is not allowed
      sal_sock = sal_get_socket((int) file->data);
packages\wiznet-latest\src\wiz_af_inet.c(77): warning:  #144-D: a value of type "int (*)(int, void *, size_t, int, struct sockaddr *, socklen_t *)" cannot be used to initialize an entity of type "int (*)(int, const struct msghdr *, int)"
      wiz_recvfrom,
packages\wiznet-latest\src\wiz_af_inet.c(78): warning:  #144-D: a value of type "int (*)(int, int, int, void *, socklen_t *)" cannot be used to initialize an entity of type "int (*)(int, struct msghdr *, int)"
      wiz_getsockopt,
packages\wiznet-latest\src\wiz_af_inet.c(79): warning:  #144-D: a value of type "int (*)(int, int, int, const void *, socklen_t)" cannot be used to initialize an entity of type "int (*)(int, void *, size_t, int, struct sockaddr *, socklen_t *)"
      wiz_setsockopt,
packages\wiznet-latest\src\wiz_af_inet.c(80): warning:  #144-D: a value of type "int (*)(int, int)" cannot be used to initialize an entity of type "int (*)(int, int, int, void *, socklen_t *)"
      wiz_shutdown,
packages\wiznet-latest\src\wiz_af_inet.c(85): warning:  #144-D: a value of type "int (*)(struct dfs_fd *, struct rt_pollreq *)" cannot be used to initialize an entity of type "int (*)(int, struct sockaddr *, socklen_t *)"
      wiz_poll,
packages\wiznet-latest\src\wiz_af_inet.c: 6 warnings, 1 error

### Other additional context

scons build error
[image]

zrz4066, 2024-10-12:
Modify the static int wiz_poll(struct dfs_fd *file, struct rt_pollreq *req)
to
static int wiz_poll(struct dfs_filesystem *file, struct rt_pollreq *req)

closed 2024-10-12 by zrz4066

CYFS3, 2024-10-12:
> Modify the static int wiz_poll(struct dfs_fd *file, struct rt_pollreq *req) to static int wiz_poll(struct dfs_filesystem *file, struct rt_pollreq *req)

#7200 I think it should be revised as dfs_fd  to dfs_file