RT-Thread/rt-thread #9534: [Feature] DM Device IDA management

GuEe-GUI, 2024-10-15:
### Describe problem solved by the proposed feature

rt_device identity: a master id - device id

```c
struct rt_device
{
    [...]
#ifdef RT_USING_DM
    rt_uint8_t                master_id;                /**< 0 - 255 */
#endif
    rt_uint8_t                device_id;                /**< 0 - 255 */
    [...]
};
```

There are some IDs:
```c
#define MASTER_ID_CUSTOM                                0

#define MASTER_ID_ADC                                   1
#define MASTER_ID_AUDIO                                 2

#define MASTER_ID_CAN                                   19

#define MASTER_ID_DAC                                   28
#define MASTER_ID_DVFS                                  29

#define MASTER_ID_HWCRYPTO                              64
#define MASTER_ID_HWTIMER                               65

#define MASTER_ID_I2C_BUS                               73
#define MASTER_ID_I2C_DEV                               74
#define MASTER_ID_INPUT                                 75

#define MASTER_ID_LED                                   102

#define MASTER_ID_MEM                                   112
#define MASTER_ID_MTD                                   113
#define MASTER_ID_MISC                                  114

#define MASTER_ID_NET                                   118
#define MASTER_ID_NVME                                  119

#define MASTER_ID_PHY                                   136
#define MASTER_ID_PIN                                   137
#define MASTER_ID_PTP                                   138
#define MASTER_ID_PWM                                   139

#define MASTER_ID_RNG                                   154
#define MASTER_ID_RPMSG_EPT                             155
#define MASTER_ID_RPMSG_CHAR                            156
#define MASTER_ID_RTC                                   157

#define MASTER_ID_SCSI_SD                               163
#define MASTER_ID_SCSI_CDROM                            164
#define MASTER_ID_SDIO                                  165
#define MASTER_ID_SENSOR                                166
#define MASTER_ID_SERIAL                                167
#define MASTER_ID_SPI_BUS                               168
#define MASTER_ID_SPI_DEV                               169

#define MASTER_ID_TEE                                   172
#define MASTER_ID_THERMAL                               173
#define MASTER_ID_TOUCH                                 174
#define MASTER_ID_TTY                                   175

#define MASTER_ID_USB_DEV                               181
#define MASTER_ID_USB_BUS                               182
#define MASTER_ID_USB_OTG                               183

#define MASTER_ID_VIDEO_BACKLIGHT                       190
#define MASTER_ID_VIDEO_FRAMEBUFFER                     191
#define MASTER_ID_VIRTUAL_BLOCK                         192

#define MASTER_ID_WATCHDOG                              199
#define MASTER_ID_WLAN                                  200
```

The IDA management API:
```c
/* ID Allocation */
struct rt_dm_ida
{
    rt_uint8_t master_id;

#define RT_DM_IDA_NUM 256
    DECLARE_BITMAP(map, RT_DM_IDA_NUM);
    struct rt_spinlock lock;
};

#define RT_DM_IDA_INIT(id)  { .master_id = MASTER_ID_##id }

int rt_dm_ida_alloc(struct rt_dm_ida *ida);
rt_bool_t rt_dm_ida_take(struct rt_dm_ida *ida, int id);
void rt_dm_ida_free(struct rt_dm_ida *ida, int id);
```

Example:
```c
static struct rt_dm_ida ptp_ida = RT_DM_IDA_INIT(PTP);

rt_err_t rt_hw_ptp_clock_register(struct rt_ptp_clock *ptp)
{
    [...]
    int device_id;
    const char *dev_name;
    char name[RT_NAME_MAX];

    if ((device_id = rt_dm_ida_alloc(&ptp_ida)) < 0)
    {
        return -RT_EFULL;
    }

    rt_dm_dev_set_name(&ptp->parent, "ptp%u", device_id);
    dev_name = rt_dm_dev_get_name(&ptp->parent);

    [...]

    ptp->parent.type = RT_Device_Class_Char;
#ifdef RT_USING_DEVICE_OPS
    ptp->parent.ops = ptp->parent.ops ? : &_ptp_ops;
#endif
    ptp->parent.master_id = ptp_ida.master_id;
    ptp->parent.device_id = device_id;

    if ((err = rt_device_register(&ptp->parent, dev_name, RT_DEVICE_FLAG_DEACTIVATE)))
    {
        goto _fail;
    }

    return RT_EOK;

_fail:
    rt_dm_ida_free(&ptp_ida, device_id);

    return err;
}

rt_err_t rt_hw_ptp_clock_unregister(struct rt_ptp_clock *ptp)
{
   [...]

    rt_dm_ida_free(&ptp_ida, ptp->parent.device_id);

    return rt_device_unregister(&ptp->parent);
}
```

Drivers can manage their own IDs without having to concern themselves with the register/unregister in system. IDA management API can used to other identity like name:

[image]
[image]

GuEe-GUI, 2024-10-16:
Update IDs:

```c
#define MASTER_ID_CUSTOM                                0

/* Block */
#define MASTER_ID_NVME                                  1
#define MASTER_ID_SCSI_SD                               2
#define MASTER_ID_SCSI_CDROM                            3
#define MASTER_ID_SDIO                                  4
#define MASTER_ID_VIRTUAL_BLOCK                         5

/* Char */
#define MASTER_ID_RPMSG_EPT                             11
#define MASTER_ID_RPMSG_CHAR                            12
#define MASTER_ID_SERIAL                                13

/* Clock Timer */
#define MASTER_ID_HWTIMER                               21
#define MASTER_ID_PTP                                   22
#define MASTER_ID_RTC                                   23

/* Graphic Display */
#define MASTER_ID_GRAPHIC_BACKLIGHT                     31
#define MASTER_ID_GRAPHIC_FRAMEBUFFER                   32
#define MASTER_ID_LED                                   33

/* Hardware Monitor */
#define MASTER_ID_DVFS                                  41
#define MASTER_ID_SENSOR                                42
#define MASTER_ID_THERMAL                               43
#define MASTER_ID_WATCHDOG                              44

/* I2C */
#define MASTER_ID_I2C_BUS                               51
#define MASTER_ID_I2C_DEV                               52

/* IO Contorl */
#define MASTER_ID_ADC                                   61
#define MASTER_ID_DAC                                   62
#define MASTER_ID_PIN                                   63
#define MASTER_ID_PWM                                   64

/* Memory */
#define MASTER_ID_MEM                                   71
#define MASTER_ID_MTD                                   72

/* MISC */
#define MASTER_ID_MISC                                  81

/* Multimedia */
#define MASTER_ID_AUDIO                                 91

/* Net */
#define MASTER_ID_CAN                                   101
#define MASTER_ID_ETH                                   102
#define MASTER_ID_PHY                                   103
#define MASTER_ID_WLAN                                  104

/* Input */
#define MASTER_ID_INPUT                                 111
#define MASTER_ID_TOUCH                                 112

/* Security */
#define MASTER_ID_HWCRYPTO                              121
#define MASTER_ID_RNG                                   122
#define MASTER_ID_TEE                                   123

/* SPI */
#define MASTER_ID_SPI_BUS                               131
#define MASTER_ID_SPI_DEV                               132

/* TTY */
#define MASTER_ID_TTY                                   141
#define MASTER_ID_PTS                                   142
#define MASTER_ID_PTY                                   143

/* USB */
#define MASTER_ID_USB_DEV                               151
#define MASTER_ID_USB_BUS                               152
#define MASTER_ID_USB_OTG                               153
```

polarvid, 2024-10-16:
话说是不是也可以把通过设备号实现的 device_find 功能加上呢？