RT-Thread/rt-thread #9536: [Bug]

Memory-afk, 2024-10-16:
### RT-Thread Version

5.1.0

### Hardware Type/Architectures

all

### Develop Toolchain

GCC

### Describe the bug

LOCAL_CPPDEFINES组下的宏定义无法添加到CMakeLists.txt文件中

WwWangGuan, 2024-11-13:
能描述的再详细一点吗？

Memory-afk, 2024-11-14:
> 能描述的再详细一点吗？
具体如图：
[image]
[image]
[image]

WwWangGuan, 2024-11-14:
ok，我去看一下

WwWangGuan, 2024-11-14:
我刚才编译了bsp/stm32f407-armfly-v5，虽然没有看到这组宏定义，但是编译没有报warning
```bash
C:\Users\Administrator\AppData\Local\Programs\CLion\bin\cmake\win\x64\bin\cmake.exe --build F:\rt-thread\bsp\stm32\stm32f407-armfly-v5\cmake-build-debug --target stm32f407-armfly-v5.elf -- -j 18
[  1%] Building C object CMakeFiles/rtt_Libraries.dir/F_/rt-thread/bsp/stm32/libraries/STM32F4xx_HAL/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rcc_ex.c.obj
[  2%] Building C object CMakeFiles/rtt_Drivers.dir/F_/rt-thread/bsp/stm32/libraries/HAL_Drivers/drivers/drv_gpio.c.obj
[  3%] Building C object CMakeFiles/rtt_Libraries.dir/F_/rt-thread/bsp/stm32/libraries/STM32F4xx_HAL/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_crc.c.obj
[  5%] Building C object CMakeFiles/rtt_DeviceDrivers.dir/F_/rt-thread/components/drivers/core/device.c.obj
[  6%] Building C object CMakeFiles/rtt_Compiler.dir/F_/rt-thread/components/libc/compilers/common/cctype.c.obj
[  7%] Building C object CMakeFiles/rtt_Drivers.dir/F_/rt-thread/bsp/stm32/libraries/HAL_Drivers/drivers/drv_spi.c.obj
[  9%] Building C object CMakeFiles/rtt_DeviceDrivers.dir/F_/rt-thread/components/drivers/ipc/completion_comm.c.obj
[ 10%] Building C object CMakeFiles/rtt_Compiler.dir/F_/rt-thread/components/libc/compilers/common/cstdlib.c.obj
[ 11%] Building C object CMakeFiles/rtt_Libraries.dir/F_/rt-thread/bsp/stm32/libraries/STM32F4xx_HAL/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rng.c.obj
[ 13%] Building C object CMakeFiles/rtt_Drivers.dir/F_/rt-thread/bsp/stm32/libraries/HAL_Drivers/drivers/drv_usart.c.obj
[ 14%] Building C object CMakeFiles/rtt_Finsh.dir/F_/rt-thread/components/finsh/cmd.c.obj
[ 15%] Building C object CMakeFiles/rtt_Libraries.dir/F_/rt-thread/bsp/stm32/libraries/STM32F4xx_HAL/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_cortex.c.obj
[ 17%] Building C object CMakeFiles/rtt_DeviceDrivers.dir/F_/rt-thread/components/drivers/ipc/completion_up.c.obj
[ 18%] Building C object CMakeFiles/rtt_Drivers.dir/F_/rt-thread/bsp/stm32/libraries/HAL_Drivers/drv_common.c.obj
[ 19%] Building C object CMakeFiles/rtt_Compiler.dir/F_/rt-thread/components/libc/compilers/common/cstring.c.obj
[ 21%] Building C object CMakeFiles/rtt_Finsh.dir/F_/rt-thread/components/finsh/msh_parse.c.obj
[ 22%] Building C object CMakeFiles/rtt_Kernel.dir/F_/rt-thread/src/clock.c.obj
[ 23%] Building C object CMakeFiles/rtt_DeviceDrivers.dir/F_/rt-thread/components/drivers/ipc/condvar.c.obj
[ 25%] Building C object CMakeFiles/rtt_libcpu.dir/F_/rt-thread/libcpu/arm/common/atomic_arm.c.obj
[ 26%] Building C object CMakeFiles/rtt_Compiler.dir/F_/rt-thread/components/libc/compilers/common/ctime.c.obj
[ 27%] Building C object CMakeFiles/rtt_libcpu.dir/F_/rt-thread/libcpu/arm/common/div0.c.obj
[ 28%] Building C object CMakeFiles/rtt_Kernel.dir/F_/rt-thread/src/components.c.obj
[ 30%] Building C object CMakeFiles/rtt_Libraries.dir/F_/rt-thread/bsp/stm32/libraries/STM32F4xx_HAL/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_spi.c.obj
[ 31%] Building ASM object CMakeFiles/rtt_Drivers.dir/F_/rt-thread/bsp/stm32/libraries/STM32F4xx_HAL/CMSIS/Device/ST/STM32F4xx/Source/Templates/gcc/startup_stm32f407xx.s.obj
[ 32%] Building C object CMakeFiles/rtt_Kernel.dir/F_/rt-thread/src/cpu_up.c.obj
[ 34%] Building C object CMakeFiles/rtt_Finsh.dir/F_/rt-thread/components/finsh/msh.c.obj
[ 35%] Building C object CMakeFiles/rtt_DeviceDrivers.dir/F_/rt-thread/components/drivers/ipc/dataqueue.c.obj
[ 36%] Building C object CMakeFiles/rtt_Compiler.dir/F_/rt-thread/components/libc/compilers/common/cunistd.c.obj
[ 38%] Building C object CMakeFiles/rtt_Libraries.dir/F_/rt-thread/bsp/stm32/libraries/STM32F4xx_HAL/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_uart.c.obj
[ 39%] Building C object CMakeFiles/rtt_Drivers.dir/board/CubeMX_Config/Src/stm32f4xx_hal_msp.c.obj
[ 40%] Building C object CMakeFiles/rtt_libcpu.dir/F_/rt-thread/libcpu/arm/common/showmem.c.obj
[ 42%] Building C object CMakeFiles/rtt_Finsh.dir/F_/rt-thread/components/finsh/shell.c.obj
[ 43%] Building C object CMakeFiles/rtt_DeviceDrivers.dir/F_/rt-thread/components/drivers/ipc/pipe.c.obj
[ 44%] Building C object CMakeFiles/rtt_Compiler.dir/F_/rt-thread/components/libc/compilers/common/cwchar.c.obj
[ 46%] Building C object CMakeFiles/rtt_Kernel.dir/F_/rt-thread/src/defunct.c.obj
[ 47%] Building C object CMakeFiles/rtt_Libraries.dir/F_/rt-thread/bsp/stm32/libraries/STM32F4xx_HAL/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rcc.c.obj
[ 48%] Building C object CMakeFiles/rtt_Drivers.dir/board/board.c.obj
[ 50%] Building ASM object CMakeFiles/rtt_libcpu.dir/F_/rt-thread/libcpu/arm/cortex-m4/context_gcc.S.obj
[ 51%] Building C object CMakeFiles/rtt_Kernel.dir/F_/rt-thread/src/idle.c.obj
[ 52%] Building C object CMakeFiles/rtt_Libraries.dir/F_/rt-thread/bsp/stm32/libraries/STM32F4xx_HAL/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_cec.c.obj
[ 53%] Building C object CMakeFiles/rtt_libcpu.dir/F_/rt-thread/libcpu/arm/cortex-m4/cpuport.c.obj
[ 55%] Building C object CMakeFiles/rtt_DeviceDrivers.dir/F_/rt-thread/components/drivers/ipc/ringblk_buf.c.obj
[ 56%] Building C object CMakeFiles/rtt_Compiler.dir/F_/rt-thread/components/libc/compilers/newlib/syscalls.c.obj
[ 57%] Building C object CMakeFiles/rtt_DeviceDrivers.dir/F_/rt-thread/components/drivers/ipc/ringbuffer.c.obj
[ 57%] Built target rtt_Finsh
[ 59%] Building C object CMakeFiles/rtt_Kernel.dir/F_/rt-thread/src/ipc.c.obj
[ 60%] Building C object CMakeFiles/rtt_Libraries.dir/F_/rt-thread/bsp/stm32/libraries/STM32F4xx_HAL/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_usart.c.obj
[... cut ...]
[ 97%] Built target rtt_Kernel
[ 98%] Building C object CMakeFiles/stm32f407-armfly-v5.elf.dir/applications/main.c.obj
[100%] Linking C executable stm32f407-armfly-v5.elf
Memory region         Used Size  Region Size  %age Used
             ROM:       86668 B         1 MB      8.27%
             RAM:        5432 B       128 KB      4.14%
   text	   data	    bss	    dec	    hex	filename
  86668	   1916	   3512	  92096	  167c0	stm32f407-armfly-v5.elf
[100%] Built target stm32f407-armfly-v5.elf

```

Memory-afk, 2024-11-14:
版本5.1.0？

WwWangGuan, 2024-11-14:
复现出来了，但是我和master的对比了一下，两边的sconscript是一样的

WwWangGuan, 2024-11-14:
看一下这个commit
[https://github.com/RT-Thread/rt-thread/commit/3e246caa1c672a5d42fb1db6620b16644a7ec58c#diff-824279aaaf391d9b0cfeaed0b83d6924d2ad895664fd8d6829ebe56bb68f8c4aR255](url)
5.1.0版本用的tools/cmake.py还是老版本的，上面的commit里面针对LOCAL_CPPDEFINES修改了，问题解决。
自己替换一下cmake.py就行了

WwWangGuan, 2024-11-27:
这个issue应该可以关掉了。

closed 2024-11-28 by Memory-afk