RT-Thread/rt-thread #9572: [Bug] at框架获取信号量超时错误判断

Bluebear233, 2024-10-23:
### RT-Thread Version

5.2

### Hardware Type/Architectures

stm32f407-atk-explorer

### Develop Toolchain

GCC

### Describe the bug

测试环境说明：使用AT组件&ESP8266软件包&RTT_OTA下载

MSH执行http_ota时候，msh线程跑在at_socket.c的at_recvfrom函数下等待sock->recv_notice信号量
[image]

同时client_parser线程接收到数据就重置sock->recv_notice信号量
[image]

导致等待信号量直接回!RT_EOK，最终数据接收超时，建议返回值改成-RT_ETIMEOUT
[image]

kurisaW, 2024-10-30:
@Bluebear233 欢迎贡献PR！

mysterywolf, 2024-11-04:
fixed：https://github.com/RT-Thread/rt-thread/pull/9615

closed 2024-11-04 by mysterywolf