RT-Thread/rt-thread #9608: [Bug] duo256m: RTT-Smart would raise `Load Page Fault` on boot

wychlw, 2024-11-02:
### RT-Thread Version

7bf8589fad1c0406ed993025c9e7055ee6725619

### Hardware Type/Architectures

milkv-duo256m-sd (SOC SG2002)

### Develop Toolchain

GCC

### Describe the bug

The RTT-Smart with default config (change the addr to assert requirement, ofcourse) would get a Load Page Fault on boot. See screen record: [image]

After some digging, seems is commit 148e577 cause this problem. After revert it, all things works again: screen record [image]

### Other additional context

I've tried different vaddr, none of them work

heyuanjie87, 2024-11-02:
参考我的bsp设置vaddr, https://github.com/heyuanjie87/bsp-MilkvDuo-xx

wychlw, 2024-11-02:
> 参考我的bsp设置vaddr, https://github.com/heyuanjie87/bsp-MilkvDuo-xx

感谢！您的的确可以在duos和duo256m上工作。但是，master依然不行…（之前设置的实际上就是 `0xFFFFFFC000200000` ）

看起来问题更复杂了…

heyuanjie87, 2024-11-03:
master上的duo riscv的驱动应该是还没支持smart的

wychlw, 2024-11-03:
嗷！原来如此。谢谢您！

shunyao666, 2024-11-05:
> master上的duo riscv的驱动应该是还没支持smart的
请问现在哪个分支已经支持了smart呢

heyuanjie87, 2024-11-06:
https://github.com/heyuanjie87/bsp-MilkvDuo-xx
https://github.com/heyuanjie87/bsp-k230
这两个仓库都可以，还有master里的qemu bsp也可以

unicornx, 2024-11-26:
#9657 已经 merge 了，请确认一下。如果没有问题请 close 这个 issue。

wychlw, 2024-11-28:
> #9657 已经 merge 了，请确认一下。如果没有问题请 close 这个 issue。

DuoS 测试无问题，手上的Duo256暂时出问题了，等待别人测试中

wychlw, 2024-11-29:
test pass, closing this issue

closed 2024-11-29 by wychlw