RT-Thread/rt-thread #9609: [Bug] Keil 编译 libcpu/arm/cortex-m33/context_rvds.S 文件出错，提示：Unknown opcode defined , expecting opcode or Macro

XYX12306, 2024-11-02:
### RT-Thread Version

5.2.0

### Hardware Type/Architectures

ra6e2-fpb

### Develop Toolchain

MDK Keil

### Describe the bug

1. 基于 master 分支最新源码；
2. 将 bsp/renesas/ra6e2-fpb/rtconfig.py 文件中的 CPU 改为：cortex-m33；
[image]

3. Env 工具执行 scons --target=mdk5 指令；
4. 使用最新 Keil 版本【v5.41.0.0】编译，libcpu/arm/cortex-m33/context_rvds.S 文件报错：Unknown opcode defined , expecting opcode or Macro
[image]

ChuanJiang0126, 2025-03-13:
keil asm has different syntax.
conditional compile like #if defined is not supported, as an alternative, IF, DEF, ELSIF, ELSE, ENDIF are used.
so, try to modify like below, it works.

	IF	:DEF:	__VFP_FP__
	IF	L:DEF:	__SOFTFP__
;#if defined (__VFP_FP__) && !defined(__SOFTFP__)
    TST     lr, #0x10                               ; if(!EXC_RETURN[4])
    VLDMFDEQ  r1!, {d8 - d15}                       ; pop FPU register s16~s31
;#endif
	ENDIF
	ENDIF

XYX12306, 2025-03-13:
> [@XYX12306](https://github.com/XYX12306) keil asm has different syntax. conditional compile like #if defined is not supported, as an alternative, IF, DEF, ELSIF, ELSE, ENDIF are used. so, try to modify like below, it works.
> 
> ```
> IF	:DEF:	__VFP_FP__
> IF	:LNOT::DEF:	__SOFTFP__
> ```
> 
> ;#if defined (**VFP_FP**) && !defined(**SOFTFP**) TST lr, #0x10 ; if(!EXC_RETURN[4]) VLDMFDEQ r1!, {d8 - d15} ; pop FPU register s16~s31 ;#endif ENDIF ENDIF
> 
> [@StackYuan](https://github.com/StackYuan) [@yandld](https://github.com/yandld)

[image]

kurisaW, 2025-06-20:
该问题已提交修复PR，先将该ISSUE状态关闭，有问题欢迎再次open

closed 2025-06-20 by kurisaW