RT-Thread/rt-thread #9622: [Feature] RT-smart 支持 duo 系列

unicornx, 2024-11-06:
### Describe problem solved by the proposed feature

目前 master 上的 duo 产品还不能跑 RT-smart，需要支持。

相关 issue 见 #9608。

Guohezu001, 2024-11-15:
[image]
刚刚测了一下，duo板子可以跑RT-smart

unicornx, 2024-11-15:
> ![image](https://private-user-images.githubusercontent.com/157772539/386432010-d91cb2e0-eeb9-4fa5-8bb0-fbb1ad2a4317.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

这不是 RT-smart，这是 标准版 RTT

RT-smart 启动的 logo 上有 Smart 字样如下：

[image]

unicornx, 2024-11-25:
> 这个 issue 可以由我来推进

#9657 已经 merge，目前 duo-S 还不支持，其他测试都正常。

unicornx, 2024-12-17:
Verified duo/duo256m/duoS with commit c1166e0bf1.

Closed this issue.

Thanks all for your excellent contribution!

closed 2024-12-17 by unicornx