RT-Thread/rt-thread #9644: [Bug] 如何在 M mode 下跑 qemu-virt64-riscv

unicornx, 2024-11-13:
### RT-Thread Version

master

### Hardware Type/Architectures

bsp/qemu-virt64-riscv

### Develop Toolchain

Other

### Describe the bug

`bsp/qemu-virt64-riscv/README_ZH.md` 上说 提供了riscv64的两种可配置运行模式，默认运行在M-Mode下。还可以运行在S-Mode 下，
并提供了相应的脚本：
- M-mode：./qemu-nographic.sh
- S-mode：./qemu-nographic-smode.sh
我对比了一下这两个脚本，没感觉有什么区别啊。
 
而且我理解现在的 qemu 默认都会走 opensbi，也就是默认只能运行 s-mode 的 RTT了（看 RTT 的配置，默认现在也是 enable 了RISCV_S_MODE， 这点和文档上说的也不一致）

所以问题是，我们还能在 QEMU 上跑 M-mode 的 RTT 吗？反正我尝试关掉了 RISCV_S_MODE 然后用 qemu 启动（用 ./qemu-nographic.sh）也是不行的。

所以感觉这个文档描述是有问题的。

unicornx, 2024-11-13:
[image]

zhangjing0303, 2024-11-13:
目前，qemu-virt64-riscv应该不支持运行在M态，编译的时候CPU相关代码使用的是libcpu/risc-v/common64/目录，线程上下文切换、中断/异常处理中直接操作的是类似ssstatus/sepc/scause这些S态的寄存器，所以直接运行M态是有问题的。

BernardXiong, 2024-11-14:
可以在代码（脚本），文档描述上移除关于M模式的内容

unicornx, 2024-11-14:
Fixed with PR #9651

closed 2024-11-14 by unicornx