RT-Thread/rt-thread #9646: [Feature] rt-smart 支持 ext4 文件系统

unicornx, 2024-11-13:
### Describe problem solved by the proposed feature

参考 <https://github.com/RT-Thread/rt-thread/issues/9623#issuecomment-2472465702> TEST 3,  由于目前 rt-smart 主线版本还未支持 ext4， 所以导致 TEST3 失败。目前可以支持 fat，所以 TEST4 成功。

希望内核能够支持 ext4。

BernardXiong, 2024-11-15:
后续会更新到ext4的软件包仓库：

https://github.com/RT-Thread-packages/lwext4

在合并新版本之前，需要对这个软件包发个版本，以能兼容之前的代码；而新的版本，希望能比较好的同时支持到DFS v1/v2版本。

unicornx, 2024-11-16:
所以这个问题的解决方案是修改 ext4 的软件包仓库代码， 即不是RTT，也不是 userapps。

@BernardXiong 那么这个 ex4 软件包仓库代码的工作有 issue 跟踪吗？我们如何知道该工作的进展以及何时完成，从而可以验证？

BernardXiong, 2024-11-17:
> 所以这个问题的解决方案是修改 ext4 的软件包仓库代码， 即不是RTT，也不是 userapps。
> 
> @BernardXiong 那么这个 ex4 软件包仓库代码的工作有 issue 跟踪吗？我们如何知道该工作的进展以及何时完成，从而可以验证？

可以在[packages/lwext4仓库的issue](https://github.com/RT-Thread-packages/lwext4/issues)上进行跟踪，也可以在这里进行跟踪。如果仅是针对这个`rt-smart支持ext4文件系统`，可以在这里进行跟踪。

unicornx, 2024-11-17:
我建议在实际修改的仓库中创建 issue 进行跟踪，所以我在 lwext4 仓库中建了一个 <https://github.com/RT-Thread-packages/lwext4/issues/26>。@BernardXiong 请 assign 并补充更多说明，如果需要的话。

这里将直接引用依赖 <https://github.com/RT-Thread-packages/lwext4/issues/26>。等其 close 后执行测试工作验证完成后即可 close 本 issue。

closed 2024-12-18 by unicornx