RT-Thread/rt-thread #9659: [Bug] Milk-v-Duo 开发板 编译RTT-Smart报错

jojoandgyc, 2024-11-17:
### RT-Thread Version

master

### Hardware Type/Architectures

cvitek

### Develop Toolchain

GCC

### Describe the bug

进入/bsp/cvitek/cv18xx_risc-v 目录，scons --menuconfig配置，scons -j16编译
[image]
编译结果如下
[image]
配置截图如下
[image]

BernardXiong, 2024-11-17:
这个本身来说就是一个编译时的代码检查，
[image]

需要对KERNEL_VADDR_START配置正确的值吧

unicornx, 2024-11-17:
请等 https://github.com/RT-Thread/rt-thread/issues/9622 这个 feature 完成。

jojoandgyc, 2024-11-17:
@unicornx 好的汪老师

jojoandgyc, 2024-11-25:
好的汪老师，拉到现在最新master版本，测试 RTT-Smart运行正常
[image]
还发现一个小问题，要手动设置几个 .sh脚本 末尾换行的格式，不然编译会报错
[image]

> > 这个本身来说就是一个编译时的代码检查， ![image](https://private-user-images.githubusercontent.com/1241087/386943069-b5e7a787-33c5-4699-b601-6f1209eebea7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI1MTk3MDksIm5iZiI6MTczMjUxOTQwOSwicGF0aCI6Ii8xMjQxMDg3LzM4Njk0MzA2OS1iNWU3YTc4Ny0zM2M1LTQ2OTktYjYwMS02ZjEyMDllZWJlYTcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMjVUMDcyMzI5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OTI4OTU5ZjMwMTdjNWJhYWYwYmFiZjRlZmQzOWIyYmVhZWIxMmVjMzhkNGQwNWYzZmU0ZjMzYTQ1NGMxM2FjOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.YcAJUoSeEbAL5cf9_A-69XVzGwUSFis44ezsZRiD5Ns)
> > 需要对KERNEL_VADDR_START配置正确的值吧
> 
> 这个问题我也遇到了，如果对大核先按照标准版编译后，再切换到 smart 版本编译，此时 enable `RT_USING_SMART` 后 `KERNEL_VADDR_START` 会恢复为默认值 0xffffffc000000000，但是 duo 上 `KERNEL_VADDR_START` 要求是 0xFFFFFFC000200000。
> 
> 所以手动在 menuconfig 中配置一下就好了。
> 
> 我感觉奇怪的是你说 “手动设置RTT-Smart 内核启动地址 ，上电启动报错”， 我这里这样修改后是好的，你要不要重新试一下？
> 
> 目前 #9657 已经 merge 了，你可以取 master 测试一下 duo。