RT-Thread/rt-thread #9671: [Bug] print format looks werid when RT_DEBUGING_AUTO_INIT is open

unicornx, 2024-11-19:
### RT-Thread Version

master

### Hardware Type/Architectures

bsp/cvitek/cv18xx_risc-v/

### Develop Toolchain

Other

### Describe the bug

enable RT_DEBUGING_AUTO_INIT 后，控制台上打印的格式比较混乱，建议改进：

```
initialize rti_board_end:0 done
......
initialize lwip_system_initlwIP-2.0.3 initialized!
:0 done
......
initialize sal_init[I/sal.skt] Socket Abstraction Layer initialize success.
:0 done
initialize utest_init[I/utest] utest is initialize success.
[I/utest] total utest testcase num: (0)
:0 done
initialize _default_console_setup[I/drivers.serial] Using /dev/ttyS0 as default console
:0 done
initialize rt_posix_stdio_init:0 done
initialize mnt_initfile system initialization done!
:0 done
initialize lwp_startup:0 done
initialize finsh_system_init:0 done
```

正常情况下，如果 init 函数中没有打印输出时，规整的输出是如下格式，例如：

```
initialize rti_board_end:0 done
```
但是一旦 init 函数中有打印输出，log 就比较混乱。譬如：
```
initialize mnt_initfile system initialization done!
:0 done
```
函数名 `mnt_init` 和打印 `file system initialization done!` 连在一起了，很容易搞混。

建议对打印统一在输出函数名后强制回车，然后另起一行打印输出结果。譬如：

```
initialize rti_board_end:
0 done
```
或者
```
initialize mnt_init
file system initialization done!
:0 done
```

这样看上去规整一些。

BernardXiong, 2024-11-20:
这块似乎没什么好办法吧，或者有什么好的方法吗？

unicornx, 2024-11-20:
> 这块似乎没什么好办法吧，或者有什么好的方法吗？

就是把打印格式改一下，看得规整点，现在内核中下面代码中的打印和 init 函数中中的打印连在一起了，加换行后至少可以分开

[image]

PR #9705 (2024-11-27): kernel: Optimize display format for auto init when debugging
Fixed #9671 

When RT_DEBUGING_AUTO_INIT is enabled, kenrel will add more print message for auto init. However, the original format easily causes the kernel printing and the init function printing to be connected together, resulting in a confusing display format.

For exmaple:
```
initialize lwip_system_initlwIP-2.0.3 initialized!
:0 done
```
Or
```
initialize sal_init[I/sal.skt] Socket Abstraction Layer initialize success.
:0 done
```

Solution: Add a carriage return to separate the kernel printing and the init function printing.

A [... cut]
closed 2024-11-27 by mysterywolf