RT-Thread/rt-thread #9681: [Feature] duo 支持在 RT-smart 上挂载 rootfs （fat 格式）

unicornx, 2024-11-20:
### Describe problem solved by the proposed feature

在 #9622  的基础上实现 duo 支持挂载 rootfs

由于目前 userapps 只支持制作 fat 格式的，所以我们第一期也先支持 fat，等 ext4 就绪了我们再支持挂载 ext4.

unicornx, 2024-11-21:
补充一些 backgroud info：

`bsp/cvitek/drivers/port/mnt_romfs.c` 这个文件是从 `bsp/bouffalo_lab/libraries/rt_drivers/port/mnt.c` 移植过来的。需要做进一步的清理，譬如 BSP_USING_ON_CHIP_FLASH 和 BSP_USING_ON_CHIP_FLASH_FS 相关的代码感觉都没有用的，可以删掉

原来你把 sd0 挂到 `/sdcard` 上感觉也没有必要。

BernardXiong, 2024-11-26:
这样后续会越来越客制化了的，可以考虑把设备树用起来，先不考虑太多外设的情况，而是先把cpu，memory，console，rootfs 等配置参数通过设备树传递给内核

unicornx, 2024-11-27:
> 这样后续会越来越客制化了的，可以考虑把设备树用起来，先不考虑太多外设的情况，而是先把cpu，memory，console，rootfs 等配置参数通过设备树传递给内核

好的，已建立长期 tracking issue #9707

PR #9703 (2024-11-27): bsp: cvitek: support mouting rootfs during bootup for RT-smart.
**注意，本 PR 包含 4 个 commits。Merge 时请不要压缩，保留这些 commits。**

Fixed #9681 

具体修改方案见 commit message。

测试结果：
duo256m 和 duo 上大小核串口显示正常。
注意测试大核时需要手动配置 enable RTC 和 SDHI 驱动。

## 拉取/合并请求描述：(PR description)

[

#### 为什么提交这份PR (why to submit this PR)

#### 你的解决方案是什么 (what is your solution)

#### 请提供验证的bsp和config (provide the config and bsp) 

- BSP:

- .config:
closed 2024-11-27 by mysterywolf