RT-Thread/rt-thread #9694: [Bug] bsp/cvitek/cv18xx_aarch64 需要维护

unicornx, 2024-11-25:
### RT-Thread Version

master

### Hardware Type/Architectures

bsp/cvitek/cv18xx_aarch64

### Develop Toolchain

GCC

### Describe the bug

bsp/cvitek 下的  cv18xx_aarch64 了？目前年久失修，如果没有人维护，我打算先将其标注为 unsupported, 如果再过段时间还是没有人出来认领的话，我就打算清理掉了

文档最好更新一下，确保使用者可以找到正确的 gcc，现在文档里没写清楚。另外 标准版和 smart 版既然都说支持了所以都需要提供一下

Z8MAN8, 2024-11-25:
我可以认领

liYony, 2024-12-10:
我这边认领

liYony, 2024-12-10:
https://github.com/RT-Thread/rt-thread/pull/9749 已在此 pr 处理

unicornx, 2024-12-25:
> #9749 已在此 pr 处理

这个 issue 涉及两个 PR：
- #9749 
- #9771

liYony, 2024-12-25:
https://github.com/RT-Thread/rt-thread/pull/9835#issuecomment-2561683801

closed 2024-12-30 by Rbb666

unicornx, 2024-12-30:
> > #9749 已在此 pr 处理
> 
> 这个 issue 涉及两个 PR：
> 
> * [[bsp][cvitek] update the arm core project #9749](https://github.com/RT-Thread/rt-thread/pull/9749)
> * [rt_hw_spinlock_t cannot be found on single core include smp_call.h #9771](https://github.com/RT-Thread/rt-thread/pull/9771)

我已经 take over 这个 issue 并且在 #9835 中解决了，所以关闭了这两个 pr。

PR #9835 (2024-12-30): bsp: cvitek: continue supporting duo 256M ARM core
**注意：为了 review 和日后 track 方便，这个 PR 包含了四个 commit，merge 时请不要压缩，保持四个 commit 即可。**

Fixed #9694。这个 PR 代替了 #9749 和 #9771，所以本 PR 合并后，这两个 PR 无需合并，可以 cancel 掉。

- patch 1: components: drivers: add RT_USING_SMP judgement inside smp_call.h
- patch 2: bsp: cvitek: use ioremap for arm64 core
- patch 3: bsp: cvitek: update combine.sh for arm64
- patch 4: bsp: cvitek: update README for cv18xx_aarch64

patch 1: cv18xx_aarch64 这个 bsp 是单核，直接在 `libcpu/aarch64/common/setup.c` 文件中 `#include <smp_call.h>` 编译会报错说找不到 `rt_hw_spinlock_t` 定义。一种直接的解决方法是修改 `libcpu/aarch64 [... cut]