RT-Thread/rt-thread #9715: [Feature] Support NXP rw612

askarigtec, 2024-11-27:
### Describe problem solved by the proposed feature

The Module is very useful. NXP RW612. it would be good if add it

BernardXiong, 2024-11-28:
Thank you for your idea.

[image]

[RW612](https://www.nxp.com/products/wireless-connectivity/wi-fi-plus-bluetooth-plus-802-15-4/wireless-mcu-with-integrated-tri-radio-1x1-wi-fi-6-plus-bluetooth-low-energy-5-4-802-15-4:RW612)

askarigtec, 2024-11-28:
thanks for your response, when do you think it will added?

BernardXiong, 2024-11-30:
> thanks for your response, when do you think it will added?

I don't know. Or you could ask NXP to do it. I've only looked into what 'rw612' means. Of course, if you finish the porting, your contribution would be greatly appreciated.

closed 2025-06-17 by kurisaW