RT-Thread/rt-thread #9760: [Feature] 验证一些应用在 milkv 的 平台上是否可以运行

unicornx, 2024-12-09:
### Describe problem solved by the proposed feature

- hello world/ Busybox init/ash/gzip（待确认）
- vim（待确认）
- dropbear（待确认）

确认这几个app在riscv上能不能跑，hello world/ Busybox init/ash/gzip/vim 在userapp中都有，dropbear后续会推出

unicornx, 2024-12-09:
有个问题，在 #8358 中这些需求是列在 milkv 下，可是这难道不是 riscv 的 common 的需求吗？和 milkv duo 有什么必然的关系呢？

[image]

polarvid, 2024-12-17:
> 有个问题，在 #8358 中这些需求是列在 milkv 下，可是这难道不是 riscv 的 common 的需求吗？和 milkv duo 有什么必然的关系呢？ [...]

这个应用验证需求来源不是验证 RV64 是否可用，而是对 milk-v 移植可靠性做 QA。

unicornx, 2025-09-02:
这个 issue 看上去没有 track 的必要了，先关掉。