RT-Thread/rt-thread #9769: [Feature] bsp/cvitek riscv 小核需要支持 cache

unicornx, 2024-12-10:
### Describe problem solved by the proposed feature

需要补充 ？？

zdtyuiop4444, 2024-12-10:
我最近正在做一个这样的应用：小核采集传感器数据发给大核，大核通过网络传输到云端。
这样小核读到的数据要想办法传到大核，直接通过mailbox传输的速度达不到要求，所以就想着使用共享内存的方式传输数据。
在共享内存传数据的时候发现有缓存一致性的问题，大核不能及时看到小核对内存的修改，反过来也一样。跟milkv的工程师交流之后发现是小核这边DCache写回问题。可以参照社区[这篇帖子](https://community.milkv.io/t/ionalloc/2952)

### **Describe problem solved by the proposed feature**
解决了大小核通过共享内存通讯时遇到的缓存一致性问题

### **Describe your preferred solution**
通过在读前写后手动flush D-Cache的方式，强制对应的内容写入到内存或直接从内存中读取

### **Describe possible alternatives**
直接关闭D-Cache, 但经测试会对性能有相当大的影响，修改之后SPI读写连5M都跑不到

unicornx, 2024-12-12:
请将你的 pr 和这个 issue 关联起来吧。然后对你的 pr 做修改后重新提交 review 即可。

具体如何操作，可以阅读：<https://github.com/plctlab/plct-rt-thread/blob/notes/0.notes/20241212-github-tips.md>

unicornx, 2025-02-05:
PR #9696 was merged. Close this issue now.

closed 2025-02-05 by unicornx