RT-Thread/rt-thread #9783: [Feature] include/rtdbg.h 中有些 debug API 可以清理了

unicornx, 2024-12-16:
### Describe problem solved by the proposed feature

- dbg_here
- dbg_enter
- dbg_exit

这三个感觉是可以直接删除了，没有人用，而且也看不出还有什么用？是历史遗迹吗？请各位看看是否可以删除？

BernardXiong, 2024-12-16:
应该都是一些便于调试使用的宏：
- `dbg_here` 会自动输出行号位置，方便没调试器的状态下来追踪代码运行情况，以确定代码已经执行过了，而不是莫名其妙挂死了；
- `dbg_enter` 应该是函数进入了；
- `dbg_exit` 应该是函数离开了；

都是一些调试时的宏。

unicornx, 2024-12-16:
> 应该都是一些便于调试使用的宏：
> 
> * `dbg_here` 会自动输出行号位置，方便没调试器的状态下来追踪代码运行情况，以确定代码已经执行过了，而不是莫名其妙挂死了；
> * `dbg_enter` 应该是函数进入了；
> * `dbg_exit` 应该是函数离开了；
> 
> 都是一些调试时的宏。

@BernardXiong 所以您的建议是还是保留这三个宏？我搜了一下已经没有人用了哎。

另外 `dbg_log` 是否值得清理一下呢？

我的想法是这些 dbg_xxx 的宏如果没有人用或者不推荐用，所以都清理掉算了，目前 rtdbg.h 这个文件看着有点冗长。

BernardXiong, 2024-12-17:
清理掉，感觉关系也不大

PR #9807 (2024-12-20): kernel: cleanup debug APIs
Fixed: #9783 

- `dbg_log` is a “NOT RECOMMENDED API”, convert the calling of this API to LOG_x and remove this API.

- `dbg_here`/`dbg_enter`/`dbg_exit`: no one use these APIs, remove them directly.

## 拉取/合并请求描述：(PR description)

[

#### 为什么提交这份PR (why to submit this PR)

#### 你的解决方案是什么 (what is your solution)

#### 请提供验证的bsp和config (provide the config and bsp) 

- BSP:

- .config:
closed 2024-12-20 by mysterywolf