RT-Thread/rt-thread #9797: [Bug] bsp/allwinner 下有一些文件是 GPL license 的，和 RTT 的 Apache license 冲突

unicornx, 2024-12-19:
### RT-Thread Version

master

### Hardware Type/Architectures

N/A

### Develop Toolchain

Other

### Describe the bug

`bsp/allwinner` 下有不少源代码含有 GPL 声明。譬如：
`bsp/allwinner/libraries/libos/include/interrupt.h`

```c
/* SPDX-License-Identifier: GPL-2.0 */
```

参考：<https://www.apache.org/licenses/GPL-compatibility.html>

> Apache 2 software can therefore be included in GPLv3 projects, because the GPLv3 license accepts our software into GPLv3 works. However, GPLv3 software cannot be included in Apache projects. The licenses are incompatible in one direction only, and it is a result of ASF's licensing philosophy and the GPLv3 authors' interpretation of copyright law.

所以我认为 Apache license 的 RT-Thread 包含了 GPL 的代码也是不恰当的，会有潜在的法律问题。

BernardXiong, 2024-12-19:
估计需要做额外声明或者把相关bsp移除了

unicornx, 2025-09-02:
too long to track, close it now

closed 2025-09-02 by unicornx