RT-Thread/rt-thread #9801: [Bug] tools/env_utility.py 包含 GPL 声明，这个和 RT-thread 的 Apache 授权冲突了

unicornx, 2024-12-19:
### RT-Thread Version

master

### Hardware Type/Architectures

N/A

### Develop Toolchain

Other

### Describe the bug

参考：<https://www.apache.org/licenses/GPL-compatibility.html>

> Apache 2 software can therefore be included in GPLv3 projects, because the GPLv3 license accepts our software into GPLv3 works. However, GPLv3 software cannot be included in Apache projects. The licenses are incompatible in one direction only, and it is a result of ASF's licensing philosophy and the GPLv3 authors' interpretation of copyright law.

所以我认为 Apache license 的 RT-Thread 包含了 GPL 的代码也是不恰当的，会有潜在的法律问题。

BernardXiong, 2024-12-19:
工具的python脚本不在这块之列，如果要更改为Apache License v2.0，需要溯源获得大家的认可。

unicornx, 2024-12-20:
> 工具的python脚本不在这块之列，如果要更改为Apache License v2.0，需要溯源获得大家的认可。

“工具的python脚本不在这块之列” 是什么意思？是说这部分不会编译进入 kernel 的 image 所以没关系？

BernardXiong, 2024-12-22:
> > 工具的python脚本不在这块之列，如果要更改为Apache License v2.0，需要溯源获得大家的认可。
> 
> “工具的python脚本不在这块之列” 是什么意思？是说这部分不会编译进入 kernel 的 image 所以没关系？

工具python代码和rt-thread本身的代码差别很大，不属于嵌入式代码的范畴的

BernardXiong, 2025-04-23:
tools下的python脚本暂时都维持GPL许可协议。python代码多是开源的，使用GPL许可协议也希望后续代码进行开源，这部分应该没什么影响。