RT-Thread/rt-thread #9812: [Question] 有没有固定的下载地址可以得到最新的可用用于开发 riscv 的 musl gcc？感觉一直没有搞清楚

unicornx, 2024-12-20:
### RT-Thread Version

master

### Hardware Type/Architectures

N/A

### Develop Toolchain

Other

### Describe the bug

以前都是参考的 .github/workflows/action_utest.yml 里写的链接。但是今天发现文件里的链接 `https://github.com/RT-Thread/toolchains-ci/releases/download/v1.7/riscv64-linux-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2` 并不能用来编译 app。

所以想知道官方的 URL。可以得到同时可以用于编译 kernel 和 app 的 musl gcc for riscv

而且最好是保持一个 latest 的 alias，就像现在  .github/workflows/action_utest.yml 里写的 `https://github.com/RT-Thread/toolchains-ci/releases/download/v1.7/riscv64-linux-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2` 一样。这样用户只要记住这个名字就好了。

BernardXiong, 2024-12-22:
@aozima 有ci build推送的链接吗？

ZhaoCake, 2024-12-30:
> 因此，即使找到了当前ci中工具链的新版本，如果没有对应的源码，依然不建议使用，毕竟如前文所述，对于ci中这个工具链是否真的启用了浮点数我保持合理的怀疑（实际上持否定态度），处理器带有fpu而不使用硬件浮点，这是不应该接受的。

我完全不能够理解，但是这个旧工具链竟然确实是编译出了带有浮点指令的程序，尽管它只能指定lp64的ABI而不能指定lp64f与lp64d。这实在是一个太出乎意料的工具链，我真是完全不能够理解。现在缺乏编译器知识的我是足够困惑的。有没有大佬能点拨一下为什么一个-mabi=lp64的工具链，在编译时接收的ABI参数也是lp64，却能够编译出带有浮点指令的程序。

话说回来，我是通过移植whetstone浮点计算基准测试确定了这一点。在QEMU中运行这种基准测试并不是好的做法，但以下结果应该仍然能够反映出结论。

rt smart：使用这个ci中的musl工具链，编译器信息如上条评论。

关闭fpu并设置rtconfig.py中-march=rv64imac, 保持-mabi=lp64

```
Loops: 100000, Iterations: 1, Duration: 103 sec.
C Converted Double Precision Whetstones: 97.1 MIPS
```

开启fpu,改回rtconfig.py的-march=rv64ima, 保持-mabi=lp64

```
Loops: 100000, Iterations: 1, Duration: 39 sec.
C Converted Double Precision Whetstones: 256.4 MIPS
```

rt-thread standard: 使用的工具链是一个挺正常的工具链，

```
❯ $RTT_EXEC_PATH/riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=/home/zhaocake/WorkSpace/Embeded/riscv64-unknown-elf-toolchain-for-rtt/bin/riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/zhaocake/WorkSpace/Embeded/riscv64-unknown-elf-toolchain-for-rtt/bin/../libexec/gcc/riscv64-unknown-elf/10.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /scratch/jenkins/workspace/tpp-freedom-tools/tpp03--build-binary-packages--parameterized/obj/x86_64-linux-ubuntu14/build/riscv64-unknown-elf-gcc/riscv-gcc/configure --target=riscv64-unknown-elf --host=x86_64-linux-gnu --prefix=/scratch/jenkins/workspace/tpp-freedom-tools/tpp03--build-binary-packages--parameterized/obj/x86_64-linux-ubuntu14/install/riscv64-unknown-elf-gcc-10.2.0-2020.12.8-x86_64-linux-ubuntu14 --with-pkgversion='SiFive GCC-Metal 10.2.0-2020.12.8' --with-bugurl=https://github.com/sifive/freedom-tools/issues --disable-shared --disable-threads --enable-languages=c,c++ --enable-tls --with-newlib --with-sysroot=/scratch/jenkins/workspace/tpp-freedom-tools/tpp03--build-binary-packages--parameterized/obj/x86_64-linux-ubuntu14/install/riscv64-unknown-elf-gcc-10.2.0-2020.12.8-x86_64-linux-ubuntu14/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=../riscv-gcc --with-system-zlib --enable-checking=yes --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc CFLAGS=-O2 CXXFLAGS=-O2 'CFLAGS_FOR_TARGET=-Os -mcmodel=medany' 'CXXFLAGS_FOR_TARGET=-Os -mcmodel=medany'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.2.0 (SiFive GCC-Metal 10.2.0-2020.12.8)
```

可以看到它的`--with-abi=lp64d`。

关闭fpu并修改rtconfig.py中-march=rv64imac， -mabi=lp64
```
Loops: 100000, Iterations: 1, Duration: 64 sec.
C Converted Double Precision Whetstones: 156.2 MIPS
```

开启fpu并修改rtconfig.py中-march=rv64imafdc, -mabi=lp64d。（如果不修改mabi理所当然地会报错，所以这个musl工具链令我头疼）

```
Loops: 100000, Iterations: 1, Duration: 26 sec.
C Converted Double Precision Whetstones: 384.6 MIPS
```

对比结果显然得出结论，该ci中的musl库工具链能够在lp64的目标ABI下编译出能够运行的带有浮点指令的程序。

我应该单独提一个issue关于这个问题。

unicornx, 2024-12-30:
@ZhaoCake 请问您是负责做工具链的吗？我提这个问题的原因是因为我发现用 `.github/workflows/action_utest.yml` 里写的 musl toolchain 链接 (<https://github.com/RT-Thread/toolchains-ci/releases/download/v1.7/riscv64-linux-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2>) 可以用于编译 kernel，但不能用来编译 app。然后我问 RTT 的同事，userapps 那边 zhujiale 和 heyuanjie 说这个链接的版本旧了。给了我另一个链接，然后就好用了，编译 kernel 和 user mode app 都没有问题。

所以我的本意是 RTT 那边是否可以发布一个统一的链接，或者把 <https://github.com/RT-Thread/toolchains-ci/releases/download/v1.7/riscv64-linux-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2> 这个链接更新到最新的版本就好。

我看您研究了很多，不知道是不是 RTT 那边已经解决了这些问题了吗？

其实我只是想要 RTT 公布一个统一的最新版本的链接给社区用。

ZhaoCake, 2024-12-31:
> 所以我的本意是希望 RTT 那边是否可以发布一个统一的链接，或者把 https://github.com/RT-Thread/toolchains-ci/releases/download/v1.7/riscv64-linux-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2 这个链接更新到最新的版本就好。

您好，我是在试图使用gdb调试qemu启动的rt smart时遇到由于工具链版本过低导致无法使用gdb调试的情况。这可以通过开启一个docker或者降低我系统版本库的方式解决。不过我看到您提出的这个issue所以尝试使用新工具链，不过显然出现了问题。

请问您可以提供新版本工具链的获取方式吗？或者更新在ci utest中。十分感谢。

unicornx, 2025-01-01:
> 请问您可以提供新版本工具链的获取方式吗？或者更新在ci utest中。十分感谢。

https://download.rt-thread.org/download/rt-smart/toolchains/riscv64gc-linux-musleabi_for_x86_64-pc-linux-gnu_222725-8a397096c1.tar.bz2

ci test 那个应该有专人维护我就不去随便动了。
我觉得最好按照我这个 issue 的建议统一到最新的以后就好了，也就是说我们只要记住形如 ”xxxx_latest.tar.bz2“ 这样的链接就好了，否则这种一大串数字的看着就很奇怪。

BernardXiong, 2025-01-01:
看下来有些迷糊，首先对rt-smart用的musl gcc工具链做些澄清，免得混淆了：

- rt-smart使用的musl gcc工具链并不是一个标准的musl gcc版本，所以无法从musl原版 + gcc原始版本构建出一份rt-smart musl gcc工具链出来；
  - 核心点是，rt-smart使用的musl是一份修改后的libc；而且这份musl libc编译出的libc_kernal.a是给内核使用的，用户态使用的则是libc.a；（所以有了kernel spec的定制）
  - gcc相关的代码应该都未进行修改，是原版的gcc工具链。核心点是musl libc上；
- rt-thread github用的ci musl gcc工具链仅用于ci build看护，所以是一个老的版本，不是新版本；
- 新版本是rt-smart内部git ci推送的，有固定的下载地址：
  - riscv版本：https://download.rt-thread.org/download/rt-smart/toolchains/riscv64gc-linux-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2
  - aarch64版本：https://download.rt-thread.org/download/rt-smart/toolchains/aarch64-linux-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2
  - 上面两个版本都是最新推送版本，所以也是非正式发布版本；理论上也包括Windows版本的工具链；

a1012112796, 2025-01-02:
> 核心点是，rt-smart使用的musl是一份修改后的libc；而且这份musl libc编译出的libc_kernal.a是给内核使用的，用户态使用的则是libc.a；（所以有了kernel spec的定制）

所以建议最好还是把名称中的目标core 直接从 `linux` 改为 `rtsmart`, 这样便于区分。例如 `aarch64-rtsmart-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2`

ZhaoCake, 2025-01-02:
感谢提供的新工具链，刚刚尝试了，已经可以用新工具链的gdb启动调试。不过这个工具链构建的时候依赖于python2.7，这在版本较新的Linux系统中是没有的，通过动态库软链接的方式可以对原系统影响尽量小地引入python2.7的依赖。解决方法如下，后续的文档或许可以用得到。

报错内容：
```
 》 $RTT_EXEC_PATH/riscv64-unknown-linux-musl-gdb rtthread.elf
/home/zhaocake/WorkSpace/Embeded/riscv64gc-linux-musl-for-rtsmart/bin/riscv64-unknown-linux-musl-gdb: error while loading shared libraries: libpython2.7.so.1.0: cannot open shared object file: No such file or directory
```

考虑使用pyenv安装一个python2.7到虚拟环境中，然后把这个环境下的动态链接库链接到`/usr/lib`或`/usr/local/lib`即可。
```bash
pyenv install 2.7.18  # pyenv的安装各个系统可能不同故略去
ln -s ~/.pyenv/versions/2.7.18/lib/libpython2.7.so.1.0 /usr/local/lib
```

然后可以正常启动gdb进行调试。

此方法适用于本身不带python2.7的系统。

unicornx, 2025-01-02:
> 感谢提供的新工具链，刚刚尝试了，已经可以用新工具链的gdb启动调试。

@ZhaoCake 请问你用的是

- 我提供的：<https://download.rt-thread.org/download/rt-smart/toolchains/riscv64gc-linux-musleabi_for_x86_64-pc-linux-gnu_222725-8a397096c1.tar.bz2>
 
OR 

- 熊大提供的 <https://download.rt-thread.org/download/rt-smart/toolchains/riscv64gc-linux-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2>

目前看起来似乎应该用熊大提供的。

ZhaoCake, 2025-01-02:
@unicornx 
> 请问你用的是

先前使用的是您提供的工具链，刚才使用了熊大提供的工具链，也同样需要`libpython2.7.so.1.0`。

但从构建日期上来看，汪老师提供的 (https://download.rt-thread.org/download/rt-smart/toolchains/riscv64gc-linux-musleabi_for_x86_64-pc-linux-gnu_222725-8a397096c1.tar.bz2) 工具链更新，不知道是不是出于稳定性的考量没有定向到latest。

https://download.rt-thread.org/download/rt-smart/toolchains/riscv64gc-linux-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2
```
  ❯❯ /home/zhaocake/Downloads/riscv64gc-linux-musleabi_for_x86_64-pc-linux-gnu/bin : ./riscv64-unknown-linux-musl-gcc -v
Using built-in specs.
COLLECT_GCC=./riscv64-unknown-linux-musl-gcc
COLLECT_LTO_WRAPPER=/home/zhaocake/Downloads/riscv64gc-linux-musleabi_for_x86_64-pc-linux-gnu/bin/../libexec/gcc/riscv64-unknown-linux-musl/10.1.0/lto-wrapper
Target: riscv64-unknown-linux-musl
Configured with: /builds/alliance/risc-v-toolchain/riscv-gcc/configure --target=riscv64-unknown-linux-musl --prefix=/builds/alliance/risc-v-toolchain/install-native/ --with-sysroot=/builds/alliance/risc-v-toolchain/install-native//riscv64-unknown-linux-musl --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++ --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=/builds/alliance/risc-v-toolchain/riscv-gcc --disable-multilib --with-abi=lp64 --with-arch=rv64imafdc --with-tune=rocket 'CFLAGS_FOR_TARGET=-O2   -mcmodel=medany -march=rv64imafdc -mabi=lp64' 'CXXFLAGS_FOR_TARGET=-O2   -mcmodel=medany -march=rv64imafdc -mabi=lp64'
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 10.1.0 (GCC)
build date: Oct 20 2023 16:21:07
build sha: 8a397096c1ef8f0e71f75edb27d7fc6996785206
build job: 547555
build pipeline: 203957
```

https://download.rt-thread.org/download/rt-smart/toolchains/riscv64gc-linux-musleabi_for_x86_64-pc-linux-gnu_222725-8a397096c1.tar.bz2
```
  ❯❯ /home/zhaocake/WorkSpace/Embeded : $RTT_EXEC_PATH/riscv64-unknown-linux-musl-gcc -v
Using built-in specs.
COLLECT_GCC=/home/zhaocake/WorkSpace/Embeded/riscv64gc-linux-musl-for-rtsmart/bin/riscv64-unknown-linux-musl-gcc
COLLECT_LTO_WRAPPER=/home/zhaocake/WorkSpace/Embeded/riscv64gc-linux-musl-for-rtsmart/bin/../libexec/gcc/riscv64-unknown-linux-musl/10.1.0/lto-wrapper
Target: riscv64-unknown-linux-musl
Configured with: /builds/alliance/risc-v-toolchain/riscv-gcc/configure --target=riscv64-unknown-linux-musl --prefix=/builds/alliance/risc-v-toolchain/install-native/ --with-sysroot=/builds/alliance/risc-v-toolchain/install-native//riscv64-unknown-linux-musl --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++ --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=/builds/alliance/risc-v-toolchain/riscv-gcc --disable-multilib --with-abi=lp64 --with-arch=rv64imafdc --with-tune=rocket 'CFLAGS_FOR_TARGET=-O2   -mcmodel=medany -march=rv64imafdc -mabi=lp64' 'CXXFLAGS_FOR_TARGET=-O2   -mcmodel=medany -march=rv64imafdc -mabi=lp64'
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 10.1.0 (GCC)
build date: May 13 2024 23:51:27
build sha: 8a397096c1ef8f0e71f75edb27d7fc6996785206
build job: 601285
build pipeline: 222725
```

unicornx, 2025-01-07:
@BernardXiong 尝试了 riscv版本：https://download.rt-thread.org/download/rt-smart/toolchains/riscv64gc-linux-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2。发现编译内核还行，但是编译 app 然后放到 ext4 文件系统下运行就会 crash

但是用 https://download.rt-thread.org/download/rt-smart/toolchains/riscv64gc-linux-musleabi_for_x86_64-pc-linux-gnu_222725-8a397096c1.tar.bz2  这个就可以编译 app。

所以那个 latest 的版本是不是还不是最好的？

BernardXiong, 2025-01-08:
原来都是这样啊，我去check下哈

BernardXiong, 2025-01-08:
还有Windows版本的工具链链接：

https://download.rt-thread.org/download/rt-smart/toolchains/aarch64-linux-musleabi_for_i686-w64-mingw32_latest.zip
https://download.rt-thread.org/download/rt-smart/toolchains/riscv64gc-linux-musleabi_for_i686-w64-mingw32_latest.zip

a1012112796, 2025-01-14:
> 所以那个 latest 的版本是不是还不是最稳定的？要不要更新一下？

可以试试这个，默认会自动下载最新的工具链 https://github.com/RT-Thread/userapps

unicornx, 2025-01-16:
> 原来都是这样啊，我去check下哈

@BernardXiong 我把这个 issue assign 给你啦

BernardXiong, 2025-02-18:
这部分是否可以结题了，后续更新到userapps或者smart-build仓库说明上：

- aarch64工具链：
  - https://download.rt-thread.org/download/rt-smart/toolchains/aarch64-linux-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2
  - https://download.rt-thread.org/download/rt-smart/toolchains/aarch64-linux-musleabi_for_i686-w64-mingw32_latest.zip
- riscv64gc工具链：
  - https://download.rt-thread.org/download/rt-smart/toolchains/riscv64gc-linux-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2
  - https://download.rt-thread.org/download/rt-smart/toolchains/riscv64gc-linux-musleabi_for_i686-w64-mingw32_latest.zip
-  arm32工具链：
   - https://download.rt-thread.org/download/rt-smart/toolchains/arm-linux-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2
   - https://download.rt-thread.org/download/rt-smart/toolchains/arm-linux-musleabi_for_i686-w64-mingw32_latest.zip

上面这些都是基于git仓库 & ci方式自动生成的最新版本工具链；后续可以按阶段性来发布版本。

unicornx, 2025-02-19:
> 这部分是否可以结题了，后续更新到userapps或者smart-build仓库说明上：

谢谢，我验证一下。

unicornx, 2025-02-24:
> * https://download.rt-thread.org/download/rt-smart/toolchains/riscv64gc-linux-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2

这个 toolchain 还是没有解决构建 user mode 程序后无法执行的问题，已经提了一个 issue
- #10028

a1012112796, 2025-03-19:
还有一点，有的人想要 rt-smart 早期版本状态所用的工具链的下载途径，目前的工具链最新状态似乎已经和老版本的代码不兼容了。

[image]

BernardXiong, 2025-05-11:
> 这部分是否可以结题了，后续更新到userapps或者smart-build仓库说明上：
>
> * aarch64工具链：
>
>   * https://download.rt-thread.org/download/rt-smart/toolchains/aarch64-linux-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2
>   * [...]

这些信息同步到smart-build仓库的[issue](https://github.com/RT-Thread/smart-build/issues/19)了，同时这些版本的工具链也提交合并到[sdk仓库](https://github.com/RT-Thread/sdk)，可以在sdk中安装后，自行简单设置一遍`RTT_CC_PREFIX`即可使用。 <smart的工具链前缀，或许后续可以在SConstruct文件中自动来指定>