RT-Thread/rt-thread #9813: [Feature] 电源管理PM，建议添加 配置选型，可以去掉 module 功能

yuqingli05, 2024-12-20:
### Describe problem solved by the proposed feature

理由如下
1、现在pm 的 module 功能，比较占用空间，和cpu资源。低功耗空闲钩子里面代码台繁琐。CPU占用高
2、module 模块主要是为了跟踪 module_id 对电源的占用，但是实际产品完全可以去掉。

### Describe your preferred solution

我自己的项目实现了一个宏开关。如下所示
[image]
这里询问下是否有相关意向，可以的话我提交PR，你们审核下

BernardXiong, 2024-12-20:
是否可以贴一些代码或你的想法，还不太清楚指的是哪块

yuqingli05, 2024-12-20:
> 是否可以贴一些代码或你的想法，还不太清楚指的是哪块
我上传一下我修改过的源码，因为限制改为txt后缀上传
components\drivers\pm\pm.c
[pm.c.txt](https://github.com/user-attachments/files/18213039/pm.c.txt)

components\drivers\include\drivers\pm.h
[pm.h.txt](https://github.com/user-attachments/files/18213062/pm.h.txt)

BernardXiong, 2024-12-22:
> > 是否可以贴一些代码或你的想法，还不太清楚指的是哪块
> 
> 我上传一下我修改过的源码，因为限制改为txt后缀上传 components\drivers\pm\pm.c [pm.c.txt](https://github.com/user-attachments/files/18213039/pm.c.txt)
> 
> components\drivers\include\drivers\pm.h [pm.h.txt](https://github.com/user-attachments/files/18213062/pm.h.txt)

可以放到自己的分支中，提PR或给链接都可以的

kurisaW, 2025-06-17:
@yuqingli05 欢迎贡献PR！