RT-Thread/rt-thread #9822: 🎯 RT-Thread 2025 Roadmap

Rbb666, 2024-12-23:
### Describe problem solved by the proposed feature

## 测试

- [ ] UTest 测试用例完善（内核、外设）
- [ ] 添加代码覆盖率（集成ci中）
- [ ] 自动打TAG
- [ ] BSP：cvitek、K230 ......

## 组件

- [ ] UORB：用于线程间 / 进程间订阅发布机制通信

## 驱动

- [ ] I3C框架
- [ ] 更多BSP适配 UART_V2框架
- [ ] 宏内核MPU的设备树适配（？）

## 文档

- [ ] 完善 Smart 文档中心
- [ ] doxygen RTT-API 文档更新

lygstate, 2024-12-23:
BSP 这么多，可以把  rt-thread-core 拆出来吗？ 只保留  libcpu, bsp 只提供最基础的几个（比如每个CPU提供一个）其它的BSP都放在 rt-thread

unicornx, 2024-12-24:
> BSP 这么多，可以把 rt-thread-core 拆出来吗？ 只保留 libcpu, bsp 只提供最基础的几个（比如每个CPU提供一个）其它的BSP都放在 rt-thread

支持。现在 bsp 下的东西太多了。建议 bsp 部分拿出来各家自己维护。RTT 仓库中只需要支持典型的 ARCH 下的几款典型的 bsp 即可，包括 qemu-virt-xxx

unicornx, 2024-12-24:
RT-Thread 的新版本发布计划以后还有吗？5.2.0 持续了 8 个月了 see a0735dcb11f0acc5e85649cfbb761bcdf68afe31

建立为 RTT 一套可预期的周期发布的机制。

unicornx, 2024-12-24:
> 文档

我对文档改进有一些想法，提了个 issue #9824，欢迎大家拍砖。

Rbb666, 2024-12-24:
> RT-Thread 的新版本发布计划以后还有吗？5.2.0 持续了 8 个月了 see [a0735dc](https://github.com/RT-Thread/rt-thread/commit/a0735dcb11f0acc5e85649cfbb761bcdf68afe31)
> 
> 建议为 RTT 建立一套 **可预期的（固定）** 周期发布的机制。

5.2.0 可以在年前发版

supperthomas, 2024-12-24:
> RTT 仓库下的 license 问题还需要清理，一部分是由 bsp 引入的，如果 bsp 能清理掉，这个问题也能一起解决。
> 
> 目前我发现的 license 冲突问题都提了 issue 了，总结如下：
> 

bsp目录下面有个md文件

unicornx, 2024-12-27:
- #8828 里的问题可以清理一下了，没有完成的可以挪到这里来继续 track。

BernardXiong, 2025-01-05:
- 对scons building.py脚本进行清理，可以更好的进行构建；
  - 清晰的模块化划分；
  - 给出Advanced User Guide Document文档，列出其中可以深入的点；

BernardXiong, 2025-01-05:
> * [🎯 RT-Thread roadmap for long term #8828](https://github.com/RT-Thread/rt-thread/issues/8828) 里的问题可以清理一下了，没有完成的可以挪到这里来继续 track。

似乎是，如果riscv相关的都完成了，这个long term基本上都完成了？！

ComerLater, 2025-02-18:
bsp是不是可以采用子模块submodule方式

lygstate, 2025-02-18:
> bsp是不是可以采用子模块submodule方式

这样不行，更复杂

StrandingNeko, 2025-02-20:
> bsp是不是可以采用子模块submodule方式

这样是可以考虑的，或者提供一个 makefile 或 yml 文件，从其他仓库拉取，以分离 rtthread 内核和  BSP

BernardXiong, 2025-04-23:
对源文件中所有[fal_cfg.h](https://github.com/RT-Thread/rt-thread/blob/ae83165609835442c25e9533fa09d2f5dc1e038a/bsp/stm32/stm32h750-artpi/board/port/fal_cfg.h#L7)GPL许可相关的进行清理，FAL组件是Apache License v2的，为什么会出现配置文件许可协议是GPL的。

BernardXiong, 2025-04-23:
> RTT 仓库下的 license 问题还需要清理，一部分是由 bsp 引入的，如果 bsp 能清理掉，这个问题也能一起解决。
> 
> 目前我发现的 license 冲突问题都提了 issue 了，总结如下：
> 
> * [[Bug] bsp/Infineon 目录下有些模块貌似声明为 GPL，会与 RT-Thread 的 Apache 授权产生冲突 #9811](https://github.com/RT-Thread/rt-thread/issues/9811)
> * [[Bug] bsp/sparkfun-redv 下有部分文件标识为 GPL，和 RT-Thread 的 Apache 授权冲突 #9810](https://github.com/RT-Thread/rt-thread/issues/9810)
> * [[Bug] bsp/stm32/ 下有部分文件标识为 GPL，和 RT-Thread 的 Apache 授权冲突 #9809](https://github.com/RT-Thread/rt-thread/issues/9809)
> * [[Bug] libcpu/ppc 下有部分文件标识为 GPL，和 RT-Thread 的 Apache 授权冲突 #9808](https://github.com/RT-Thread/rt-thread/issues/9808)
> * [[Bug] tools/env_utility.py 包含 GPL 声明，这个和 RT-thread 的 Apache 授权冲突了 #9801](https://github.com/RT-Thread/rt-thread/issues/9801)
> * [[Bug] bsp/nuvoton/libraries/nu_packages/DA9062/da9062_reg.h 是 GPL 声明，和 RT-Thread 的 Apache 授权冲突 #9799](https://github.com/RT-Thread/rt-thread/issues/9799)
> * [[Bug] bsp/allwinner 下有一些文件是 GPL license 的，和 RT-Thread 的 Apache 授权冲突 #9797](https://github.com/RT-Thread/rt-thread/issues/9797)
> * [[Bug] examples/utest/testcases/posix 下部分代码是 GPL 的，和 RT-Thread 的 Apache 授权冲突 #9796](https://github.com/RT-Thread/rt-thread/issues/9796)
> 
> 另外设备树部分借用了很多外部的代码，是 GPL 和 BSD 双 license 的，我不确定对于这种情况，RTT 是不是要加个 NOTICE 声明，虽然 Apache 项目可以包含 BSD 的，这个可能需要专业法律回答一下。

tools下的python脚本暂时都维持GPL许可协议。python代码多是开源的，使用GPL许可协议也希望后续代码进行开源，这部分应该没什么影响。

lygstate, 2025-09-24:
> bsp是不是可以采用子模块submodule方式

其它内容不变，可以把kernel 和components 拆分出去，然后用 subtree的方式同步回来，参考 rust 对 一些依赖的处理