RT-Thread/rt-thread #9824: [Feature] RTT doxygen 文档改进

unicornx, 2024-12-23:
### Describe problem solved by the proposed feature

这个 issue 用于搜集 RTT 代码仓库中和 doxygen 文档维护改进相关的问题：

unicornx, 2024-12-24:
有关文档改进的工作我也很有兴趣，希望能在相关方面做些工作，推进一下。

BernardXiong, 2024-12-24:
文档中心

[image]

BernardXiong, 2024-12-24:
英文documentation的部署是在这里： https://www.rt-thread.io/document/site/

不清楚是否是机器人自动化部署的了，可能不是。

supperthomas, 2024-12-24:
#9424

unicornx, 2024-12-24:
> * [[Feature] [doc] doxygen文档中缺少driver相关的doxygen #9424](https://github.com/RT-Thread/rt-thread/issues/9424)
> 
> 这个doxygen还有很多未更新的地方，汪老师可以PR维护一下。

谢谢补充，我更新到 issue 的 问题 2 中了

unicornx, 2024-12-27:
本文档用于描述 API 文档的层次结构设计

```
- Basic Definitions
- Device Virtual File System
- finsh shell
- Hardware Related Package
- RT-Thread Kernel API
  - Thread Management
  - Clock and Timer Management
  - Kernel Object Management
  - Inter-Thread Communication
    - Semaphore
    - Mutex
    - Event
    - Mailbox
    - Messagequeue
  - Signal
  - Memory Management
  - Device System
  - Runtime Trace and Record
  - Other useful kernel service
  - Error Code
- Application Module
- System Initialization
- RTTHREAD Driver
  - ADC
  - DAC
  - ...
 ```

unicornx, 2025-01-06:
部分涉及文档中心改进的内容我新建了一个 #9871 来跟进，这个 issue 以后将主要用于搜集和跟进代码(.h/.c）文件中 doxygen 编写 API 文档的工作。

1078249029, 2025-01-19:
> * [[Feature] [doc] doxygen文档中缺少driver相关的doxygen #9424](https://github.com/RT-Thread/rt-thread/issues/9424)
> 
> 
> 这个doxygen还有很多未更新的地方，汪老师可以PR维护一下。

请问这个还需要补充doxygen么？我看到熊大在https://github.com/RT-Thread/rt-thread/pull/9924#issuecomment-2599753468说最好先梳理lwp模块，之后再补充doxygen注释。driver是否也有类似的问题？

supperthomas, 2025-01-19:
[image]

http://rt-thread.github.io/rt-thread/group___drivers.html

我觉得这部分应该还是不是很完善的。

参考下面链接中的文件

https://github.com/RT-Thread/rt-thread/tree/master/components/drivers
https://github.com/RT-Thread/rt-thread/tree/master/components/drivers/include/drivers

BernardXiong, 2025-01-30:
以下几部分的文档可以稍微暂缓些的，因为这块的代码大概率是需要进行梳理，甚至重构：
- components/lwp的代码，即smart的代码，整体是相对混乱的；
  - vdso需要加入多架构的设计；
- components/drivers/hwtimer、components/drivers/ktime、components/drivers/cputime的代码，各种time/timer，缺乏一个完整的设计；

supperthomas, 2025-02-21:
https://jothepro.github.io/doxygen-awesome-css/
这个doxygen挺漂亮的。

supperthomas, 2025-04-01:
看到一个中英文的readthedoc，可以参考一下。

https://github.com/hpmicro/hpm_sdk/tree/main/docs

英文
https://hpm-sdk.readthedocs.io/en/latest/
中文
https://hpm-sdk.readthedocs.io/zh-cn/latest/

[image]

仅供参考，没有其他建议。