RT-Thread/rt-thread #9833: [Bug]

godmial, 2024-12-25:
### RT-Thread Version

master

### Hardware Type/Architectures

gd32470z-lckfb

### Develop Toolchain

GCC

### Describe the bug

未对BSP做任何修改的前提下，使用scons命令生成的keil工程编译报错
[image]

报错定位
[image]

屏蔽这个结构体成员之后编译可以通过
[image]

godmial, 2024-12-25:
[image]

已提交PR

unicornx, 2024-12-25:
请把 issue 的 title 补充完整

Rbb666, 2024-12-27:
> 
> [image]
> 
> 已提交PR

请在此issue或pr中互相关联下

godmial, 2024-12-27:
#9834

PR #9834 (2024-12-27): 解决了components\libc\compilers\common\include\dirent.h libc_dirent结构体报错的bug
## 拉取/合并请求描述：(PR description)

#### 为什么提交这份PR (why to submit this PR)
使用gd32470z-lckfb BSP时发现，在未经任何修改的前提下，使用scons命令生成的keil工程编译不通过
[image]

BUG原因：components\libc\compilers\common\include\dirent.h文件中，RT_USING_MUSLLIBC 宏的使用使得 ino_t 类型的定义依赖于该宏是否被定义。如果该宏未定义，ino_t 类型将未声明，导致编译错误。
[image]

#### 你的解决方案是什么 (what is your solution)
将ino_t d_ino;也加入条件编译
[image]
 resolved #9833 
- BSP:bsp\gd32\arm\gd32470z-lckfb

- .config:无

- action:
[image]
]

### 当前拉取/合并请求的状态 Intent for your PR

必须选择一项 Choose one (Mandatory):

- [ ] 本拉取/合并请求是一个草稿版 [... cut]
closed 2024-12-27 by supperthomas