RT-Thread/rt-thread #9839: [Feature] RT_USING_NEWLIBC/RT_USING_LIBC 是否可以清理一下？

unicornx, 2024-12-27:
### Describe problem solved by the proposed feature

代码中存在一些地方对 RT_USING_NEWLIBC/RT_USING_LIBC 的判断，但是在 Kconfig 中找不到它们的定义了

这个历史是什么，谁可以讲一讲？

以及这些配置是否还有存在的意义？是否可以清理掉？

ZhaoCake, 2024-12-30:
我个人不建议清理这两个宏。在Kconfig中实际上有选项间接作用于它。这个宏共同作用是控制是否启用newlib。

首先是这两个宏的定义位置：`components/libc/compilers/newlib/Sconscript`
```python
if newlib_version and not GetDepend('RT_USING_EXTERNAL_LIBC'):
    print('Newlib version: ' + newlib_version)

    cwd = GetCurrentDir()
    src = Glob('*.c')

    CPPPATH = [cwd]
    CPPDEFINES = ['RT_USING_NEWLIBC', 'RT_USING_LIBC', '_POSIX_C_SOURCE=1'] # identify this is Newlib, and only enable POSIX.1-1990
    LIBS = ['c', 'm'] # link libc and libm
    AddDepend(['RT_USING_NEWLIBC', 'RT_USING_LIBC'])

    group = group + DefineGroup('Compiler', src, depend = [''], CPPPATH = CPPPATH, CPPDEFINES = CPPDEFINES, LIBS = LIBS)

    list = os.listdir(cwd)
    for d in list:
        path = os.path.join(cwd, d)
        if os.path.isfile(os.path.join(path, 'SConscript')):
            group = group + SConscript(os.path.join(d, 'SConscript'))
```

可以获知当存在指定版本的newlib并且没有设置`RT_USING_EXTERNAL_LIBC`配置选项为真的时候，将会定义'RT_USING_NEWLIBC', 'RT_USING_LIBC', '_POSIX_C_SOURCE=1'。

而`RT_USING_EXTERNAL_LIBC`的出现在Kconfig中，尽管不是一个用户可从menuconfig中访问的选项：`components/libc/Kconfig`
```
menu "C/C++ and POSIX layer"

config RT_USING_EXTERNAL_LIBC
    bool
    help
        This is for external libc(e.g. mlib),
        and NOT for newlib or picolibc which are inherent in the toolchains.
        Usually, the external libc should be a software package and select
        RT_USING_EXTERNAL_LIBC in software package's Kconfig
        This option is not available for users to select.

rsource "compilers/common/Kconfig"
rsource "posix/Kconfig"
rsource "cplusplus/Kconfig"

endmenu
```

从该配置选项help可以获知，这个选项应该是当使用外部C库的时候在C库软件包中直接配置的，因此不被用户直接选择。

unicornx, 2024-12-30:
@ZhaoCake 了解了，非常感谢

close 这个 issue.