RT-Thread/rt-thread #9845: [Question] RTT 遵循的 POSIX 规范版本是多少？

unicornx, 2024-12-30:
### RT-Thread Version

master

### Hardware Type/Architectures

N/A

### Develop Toolchain

Other

### Describe the bug

阅读了 <https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-standard/programming-manual/libc/introduction?id=_3-posix-%e6%a0%87%e5%87%86> 和代码。

但是看不出来 RTT 目前支持的 POSIX.1 (IEEE 1003.1) 是哪个版本？2001/2008/... 还是最新的？

建议固定明确一个，这样后面做 utest 的时候也可以有明确的目标。

另外有关对 1003.13 的支持情况，也想问问 RTT 是遵循的哪个版本？

unicornx, 2024-12-31:
> 2008

要不先固定为 2008？定下来也好衡量测试用例的标准。

另外发现 PSE 的标准，也就是 POSIX 1003.12 的文本不是免费的，这个哪里可以获得呢？

BernardXiong, 2025-01-01:
感觉可能更偏2003标准，

https://posix.opengroup.org/certification_guide.html

[image]

unicornx, 2025-01-02:
> 感觉可能更偏2003标准，

看了一下 https://posix.opengroup.org/

"1003.1™-2016 Base" 和 "1003.1™-2003 Base" 看上去是两个认证项目的名字，它们对应的 POSIX 标准其实分别是：
- IEEE Std 1003.1,2016 Edition，也就是 POSIX.1-2008
- EEE Std 1003.1-2001，也就是 POSIX.1-2001

所以如果是 “更偏2003标准”， 则实际对应的是 POSIX.1-2001 ?

BernardXiong, 2025-01-03:
> > 感觉可能更偏2003标准，
> 
> 看了一下 https://posix.opengroup.org/
> 
> "1003.1™-2016 Base" 和 "1003.1™-2003 Base" 看上去是两个认证项目的名字，它们和 POSIX 标准的对应关系分别是：
> 
> * "1003.1™-2016 Base" ---> `IEEE Std 1003.1,2016 Edition`，也就是 POSIX.1-2008
> * "1003.1™-2003 Base" ---> `IEEE Std 1003.1-2001`，也就是 POSIX.1-2001
> 
> 所以如果是 “更偏2003标准”， 则实际对应的是 POSIX.1-2001 ?

可能是呢

unicornx, 2025-01-03:
> > 
> > 所以如果是 “更偏2003标准”， 则实际对应的是 POSIX.1-2001 ?
> 
> 可能是呢

建议对 RTT 的 POSIX compliance 问题用比较正式的文档写出来，这也是我觉得在文档上值得改进和补充的地方。
可以参考一下 RTEMS 对 POSIX 的文档：
- RTEMS POSIX API Guide ：<https://docs.rtems.org/docs/main/posix-users.html>
- RTEMS POSIX 1003.1 Compliance Guide: <https://docs.rtems.org/docs/main/posix-compliance.html>