RT-Thread/rt-thread #9858: [Bug] Potential Vulnerability of Arbitrary Memory Write/Read in the devmem

x-codingman, 2025-01-02:
### RT-Thread Version

v5.1.0

### Hardware Type/Architectures

m460, n9h30 and ma35 in nuvoton 

### Develop Toolchain

Other

### Describe the bug

I am reaching out to report a potential vulnerability identified in the latest version of RT-Thread. I open this issue for your review. Kindly inform me if you plan to request a CVE ID upon confirmation of the vulnerability. I am happy to provide any additional details or clarification as needed.

## Summary

A critical arbitrary memory read/write vulnerability has been identified in the RT-Thread devmem command implementation. This vulnerability allows an attacker to read from and write to arbitrary physical memory addresses without proper access controls or validation.

## Details

### Vulnerable Code Location

The vulnerability exists in several platforms including m460, n9h30, and ma35 systems. For example, in the m460 implementation located at bsp/nuvoton/libraries/m460/rtt_port/drv_common.c, the function accepts user-controlled memory addresses and values without proper validation, and directly dereferences these addresses for memory operations.

```
void devmem(int argc, char *argv[])
{
    volatile unsigned int u32Addr;
    unsigned int value = 0, mode = 0;

    if (argc < 2 || argc > 3)
    {
        goto exit_devmem;
    }

    if (argc == 3)
    {
        if (rt_sscanf(argv[2], "0x%x", &value) != 1)
            goto exit_devmem;
        mode = 1; //Write
    }

    if (rt_sscanf(argv[1], "0x%x", &u32Addr) != 1)
        goto exit_devmem;
    else if (!u32Addr || u32Addr & (4 - 1))
        goto exit_devmem;

    if (mode)
    {
        *((volatile uint32_t *)u32Addr) = value; //vul of arbibrary memory write
    }
    rt_kprintf("0x%08x\n", *((volatile uint32_t *)u32Addr)); //vul of arbibrary memory read

    return;
exit_devmem:
    rt_kprintf("Read: devmem <physical address in hex>\n");
    rt_kprintf("Write: devmem <physical address in hex> <value in hex format>\n");
    return;
}
MSH_CMD_EXPORT(devmem, dump device registers);
```

### Vulnerability Description

The function recieves inputs from user. However, it allows direct memory access through pointer dereferencing without any input validation.

## Impact

This vulnerability exists because the function accepts user-controlled memory addresses and values without proper validation, and directly dereferences these addresses for memory operations. An attacker can exploit this vulnerability by providing arbitrary physical addresses to read sensitive information from any memory location or write malicious values to compromise the system.

BernardXiong, 2025-01-02:
Hello, 
Thank you for your feedback. These issues are related with nuvoton bsp. Please submit this issue in [nuvoton](https://github.com/OpenNuvoton).  Thank you.

x-codingman, 2025-01-03:
> Hello, Thank you for your feedback. These issues are related with nuvoton bsp. Please submit this issue in [nuvoton](https://github.com/OpenNuvoton). Thank you.

This is the portable code for Nuvoton targeting RT-Thread, as I could not locate the same code in Nuvoton's official [repository](https://github.com/OpenNuvoton). I believe these functions present a significant security risk, as they may directly accept malicious inputs. They should not be exposed to users. It would be beneficial to include a warning or cautionary note to alert developers to the potential dangers.

BernardXiong, 2025-01-03:
> devmem

maybe here?

https://github.com/OpenNuvoton/LVGL_NUMAKER-HMI-N9H30/blob/b80b7836225bd5ee7013344653700b4deede2798/libraries/n9h30/rtt_port/drv_common.c#L81

x-codingman, 2025-01-03:
> > devmem
> 
> maybe here?
> 
> https://github.com/OpenNuvoton/LVGL_NUMAKER-HMI-N9H30/blob/b80b7836225bd5ee7013344653700b4deede2798/libraries/n9h30/rtt_port/drv_common.c#L81

Thank you! This is exactly the devmem implementation for the Nuvoton N9H30. I have already reported this issue to Nuvoton for this BSP. However, for the M460, I was unable to locate the rtt_port code in Nuvoton’s repository (https://github.com/OpenNuvoton/m460bsp). Is this portable code developed by both Nuvoton and RT-Thread contributors, as indicated in the commit history?

x-codingman, 2025-01-07:
Hi there, [Nuvoton]("https://github.com/OpenNuvoton/LVGL_NUMAKER-HMI-N9H30/issues/1") has responded to this issue and will be adding appropriate warning messages to remind users to exercise caution when using this command. As standard practice, I plan to request a CVE ID for every confirmed vulnerability. Thanks.

closed 2025-01-08 by BernardXiong