RT-Thread/rt-thread #9864: [Bug] scons --target=vsc 排除文件过多问题

StrandingNeko, 2025-01-04:
### RT-Thread Version

5.1.x

### Hardware Type/Architectures

hpm5300evk

### Develop Toolchain

Microsoft VScode

### Describe the bug

生成的workspace排除的文件过多。

[image]

StrandingNeko, 2025-01-04:
能否生成与 RTT Studio 一样的文件格式，或者给一些指导，我来修改也可以。

supperthomas, 2025-01-04:
> 能否生成与 RTT Studio 一样的文件格式，或者给一些指导，我来修改也可以。

tools下面的vsc.py文件，具体你希望怎样呢？排除的文件夹有没有编译到呢？

supperthomas, 2025-01-04:
目前没有编译到的文件夹都会排除

StrandingNeko, 2025-01-04:
> 目前没有编译到的文件夹都会排除

但我在 hpm5300 evk 上的测试是，packages 的文件夹会被排除，但实际是编译进来了

StrandingNeko, 2025-01-04:
比如现在，观察VSCode，packages 的文件夹被排除，也无法被索引到，但scons 是可以编译进来对应的文件夹的

[image]
[image]

supperthomas, 2025-01-04:
> 比如现在，观察VSCode，packages 的文件夹被排除，也无法被索引到，但scons 是可以编译进来对应的文件夹的
> 
> [image]
> [image]
> 

你看看json文件里面有没有

StrandingNeko, 2025-01-04:
bsp文件夹下的 compile-commands.json 文件没有 package 路径

StrandingNeko, 2025-01-04:
build 文件夹下的有 packages 路径

supperthomas, 2025-01-04:
> bsp文件夹下的 compile-commands.json 文件没有 package 路径

那看来还是scons --cdb位置不对，你根据那个pr再往前挪挪看

StrandingNeko, 2025-01-04:
> > bsp文件夹下的 compile-commands.json 文件没有 package 路径
> 
> 那看来还是scons --cdb位置不对，你根据那个pr再往前挪挪看

不是在对应板子的 BSP 下么

StrandingNeko, 2025-01-04:
> > bsp文件夹下的 compile-commands.json 文件没有 package 路径
> 
> 那看来还是scons --cdb位置不对，你根据那个pr再往前挪挪看

我先尝试一下

StrandingNeko, 2025-01-04:
> > 能否生成与 RTT Studio 一样的文件格式，或者给一些指导，我来修改也可以。
> 
> tools下面的vsc.py文件，具体你希望怎样呢？排除的文件夹有没有编译到呢？ 后面可能考虑bsp下面的文件夹不排除

bsp下面的文件夹不排除是个好主意，我发现是我bsp文件夹下的 compile-commands.json 文件导致了这个问题，后续会跟踪一下这个问题，感谢

StrandingNeko, 2025-01-04:
此外，我觉得将 BSP 下的文件夹提到与源码同一级目录下也会更匹配 rtt studio 中的文件结构。
[image]

closed 2025-01-04 by StrandingNeko

supperthomas, 2025-01-04:
> 此外，我觉得将 BSP 下的文件夹提到与源码同一级目录下也会更匹配 rtt studio 中的文件结构。
> [image]
> 

好的，感谢反馈，后面持续改进