RT-Thread/rt-thread #9881: [Bug] pthread_cancel线程无法退出

wycwyhwyq, 2025-01-07:
### RT-Thread Version

c9c103f30e28b31e5d393e7a8cc1a07779fb99dc

### Hardware Type/Architectures

risc-v c908

### Develop Toolchain

GCC

### Describe the bug

调用pthread_cancel线程无法退出，附上测试代码
```c
#include <stdio.h>
#include <stdlib.h>
#include <stdint.h>
#include <pthread.h>
#include <unistd.h>
#include <errno.h>
#include <time.h>

pthread_t tid;

static void *subthread(void *arg)
{
    while (1) {
        usleep(1000000);
        printf("%s: %d\n", __func__, __LINE__);
    }

    return 0;
}

int main(int argc, char **argv)
{
    pthread_create(&tid, NULL, subthread, 0);

    while (getchar() != 'q')
        usleep(10000);

    printf("%s: %d\n", __func__, __LINE__);
    pthread_cancel(tid);
    printf("%s: %d\n", __func__, __LINE__);
    pthread_join(tid, NULL);
    printf("%s: %d\n", __func__, __LINE__);

    return 0;
}

```

unicornx, 2025-01-07:
什么板子上发现的？

wycwyhwyq, 2025-01-09:
> 什么板子上发现的？

CanMV-K230

unicornx, 2025-01-09:
> > 什么板子上发现的？
> 
> CanMV-K230

你用的什么 bsp 在测试？是 bsp/k230 吗？我怎么听说这个 bsp 在 canMV 上跑不起来的？

wycwyhwyq, 2025-01-10:
> > > 什么板子上发现的？
> > 
> > 
> > CanMV-K230
> 
> 你用的什么 bsp 在测试？是 bsp/k230 吗？我怎么听说这个 bsp 在 canMV 上跑不起来的？

bsp/k230系列的板子都可以复现，可能和板子没有关系，我定位到问题大概是在musl lib的cancel信号处理部分
[image]
这里会导致被cancel的线程一直接收到信号

heyuanjie87, 2025-01-10:
通过qemu调试用户态发现pthread_kill这个函数一直被调用，看c库代码里是没有循环调用的

heyuanjie87, 2025-01-10:
编译c库的时候关闭优化(-O0)pthread_cancel能正常退出，但pthread_join无法退出(被挂起在了pthread->detach_state的futex上)

eatvector, 2025-07-09:
> 编译c库的时候关闭优化(-O0)pthread_cancel能正常退出，但pthread_join无法退出(被挂起在了pthread->detach_state的futex上)

你好，请问对应的c库源码在哪里找啊，我最近在尝试解决这个问题，但是userapp xmake下载下来的编好的musl库没有调试信息

rcitach, 2025-08-05:
测试代码：
~~~c
#include <stdio.h>
#include <stdlib.h>
#include <stdint.h>
#include <pthread.h>
#include <unistd.h>
#include <errno.h>
#include <time.h>

pthread_t tid;

static void *subthread(void *arg)
{
    while (1) {
        printf("%s: %d\n", __func__, __LINE__);
        usleep(1000);
    }

    return 0;
}

int main(int argc, char **argv)
{
    pthread_create(&tid, NULL, subthread, 0);

    usleep(100);

    printf("%s: %d\n", __func__, __LINE__);
    pthread_cancel(tid);
    printf("%s: %d\n", __func__, __LINE__);
    pthread_join(tid, NULL);
    printf("%s: %d\n", __func__, __LINE__);

    return 0;
}
~~~

k230 01stdio板卡运行结果

[image]

rcitach, 2025-08-05:
# 测试流程

##  一、SMART Kernel 编译

使用最新版本工具链，根据[README.md](https://github.com/RT-Thread/rt-thread/blob/master/bsp/k230/README.md) 编译bsp/k230   
> 注：需要在menuconfig调整 RT_NAME_MAX 的大小，并在 `FPU precision`选择` Software floating-point`
[image]

[image]

## 二、生成img镜像
同样根据 [README.md](https://github.com/RT-Thread/rt-thread/blob/master/bsp/k230/README.md)  生成 K230 RTOS Only SDK的基础镜像 并更新 RT-Thread 内核

## 三、添加用户态测试程序

在过程二中，会将镜像烧录到sd卡中，再将sd卡插入到电脑中，可以发现会多出 BIN SDCARD两个盘符

[image]

使用工具链编译上述测试代码（以pthread.c命令）：
~~~shell
/opt/riscv64-linux-musleabi_for_x86_64-pc-linux-gnu/bin/riscv64-linux-musleabi-gcc pthread.c  --static -o pthread_cancel
~~~
将编译出的pthread_cancel复制到 SDCARD盘符中

## 四、运行用户态测试程序

将SD卡插入板卡中，上电复位，在控制台运行 `pthread_cancel`

[image]